SmokeMonsterPacks/Super-NT-Jailbreak #1: Height and Width bug

SegaSnatcher, 2018-02-14:
After loading a couple of roms the width and height will default to stock settings.   Once I go back to the width and height option it then automatically reverts back to my custom settings.

ghogan42, 2018-02-14:
I've seen something similar with official firmware (3.9 or 4.0) when switching between the cart and Turrican when I was changing video modes and settings.  I didn't try to reproduce it. But the official firmware might have a similar/same bug.

swt83, 2018-02-14:
I got mine yesterday, haven't tried this mod yet, and have resolution issues right off the bat.  The menu is cropped off the top so I can't see all my options.  Some games play fine, some games are cropped off the top.

anpage, 2018-02-14:
@swt83 This is a bad place to discuss this, but try adjusting overscan or "picture size" on your TV. If you still have problems, reach out to Analogue's support.

swt83, 2018-02-15:
My issue was solved by simply pressing the "pic size" button on my remote.

SegaSnatcher, 2018-02-15:
> 
I've seen something similar with official firmware (3.9 or 4.0) when switching between the cart and Turrican when I was changing video modes and settings. I didn't try to reproduce it. But the official firmware might have a similar/same bug. 

Well, I wasn't messing with video modes or settings when this happened.  I'll simply play a couple of roms and randomly the default size settings revert back to default.

scotthammack, 2018-02-15:
I've encountered this bug also. Loading a ROM seemed to ignore my custom width/height settings. Going into the width/height menu, I could see that the numbers were still set to my desired values, but the image on screen did not reflect those values. But as soon as I changed one of them, the image suddenly popped to the desired size.

iancourt, 2018-02-15:
I've had similar with the scanlines. Setting enabled, but not shown. The moment I alter another video setting (e.g. gamma boost), the scanlines suddenly show.

rgdelato, 2018-02-15:
> I've had similar with the scanlines. Setting enabled, but not shown. The moment I alter another video setting (e.g. gamma boost), the scanlines suddenly show.
I've also seen this same issue with the Turrican games using the official firmware, so it's likely not an issue with the jailbreak itself.

romevi, 2018-02-17:
I've had this happen a handful of times. For me I just turn on/off the Advance Video setting and it automatically adjusts the correct video setting. 
An annoying but easily fixable hiccup.

cacophony1, 2018-02-17:
I can confirm that this happens 100% of the time after a fresh boot of the console. My prior saved custom width/height settings aren't reflected until I modify one of the values by 1 pixel, at which point my prior setting seems to immediately load.

chipperkwah, 2018-02-25:
This is happening with me, exactly as described above! I'm sure kevtris is already fixing it...

SmokeMonsterPacks, 2018-03-04:
This should be addressed in v4.4

closed 2018-03-04 by SmokeMonsterPacks

SegaSnatcher, 2018-03-04:
Hip Hop Hurray!!