SmokeMonsterPacks/Super-NT-Jailbreak #2: SRAM saving isn't working correctly

Aeana, 2018-02-14:
I haven't been able to get any game that uses SRAM to keep its save consistently.  Every once in a while, switching games and returning to the prior game will show me the save I created, but most of the time the save is corrupted.  This is easily verifiable with games that produce an error sound when they try to read corrupted SRAM, such as Dragon Quest 5, Dragon Quest 6, or Glory of Heracles 3.

Surefire way to reproduce:

Start Dragon Quest 5, and create a file. As soon as name entry is complete and you see the game start, open the file menu and select another game that also writes to SRAM. In this case, try Dragon Quest 6.
Do the same thing with Dragon Quest 6: do the name entry and create a file, then go back to the file menu and load Dragon Quest 5 again.

Error sound, save corrupted.

xanfir, 2018-02-14:
I am also experiencing the random save deletion and/or corruption with SRAM games. Even after going back to the file menu.

Aeana, 2018-02-15:
I've done a bit more experimentation with this... 

Using the advice given in the other issue about using "SD Card Formatter," I did that, and now I have a different set of results. I've been able to reliably save and restore saves on Dragon Quest 5 and Emerald Dragon through all kinds of switches. But Dragon Quest 3, Dragon Quest 6 (even JP without the janky English patch), and Glory of Heracles 3 will still very frequently lose their saves.

I wish I could pinpoint more specific behavior to be of more help. I'll keep testing different scenarios and reporting back if I find anything interesting.

SmokeMonsterPacks, 2018-02-15:
I read of someone who [fixed their save problems](https://www.resetera.com/posts/4553337/) by formatting their SD card using [SD Formatter 5](https://www.sdcard.org/downloads/formatter_4/). Please test this and report back. Although I'm not sure if that tool supports formatting >32GB cards to FAT32. If that is the case, please test [fat32format](http://www.ridgecrop.demon.co.uk/index.htm?guiformat.htm). 

This is urgent and help testing solutions is greatly appreciated.

Aeana, 2018-02-15:
I personally did try that, as noted in my above post. It seems to help a little, but eventually the same problems arise.  The same thing goes for the person whose post you linked, unfortunately. Here's a later post from the same person: https://www.resetera.com/posts/4559248/

seventeenbells, 2018-02-15:
Yeah like Aeana said the save stopped working for me as well but formatting with that formatter did at least make it last longer (I loaded a few other games and it was fine). Any steps you would suggest taking specifically to test specifically? I’m going to try tinkering with file names and other sd cards I have lying around and will report back a little later today.

axebox, 2018-02-15:
Saves didn't work for me for any game. I formatted the card using OSX Disk Utility as FAT. The physical card is a high speed SanDisk microsd with an SD adapter.

danhermle, 2018-02-15:
I just found an old 16GB micro (was using a 64GB formatted with GUIFormat since Windows won’t do > 32GB FAT32) and the result is the same.  Saves don’t work.  I used Windows to format the 16GB card this time thinking maybe it was the tool’s fault but that didn’t affect the result.

danhermle, 2018-02-15:
Setting the SD card speed to Normal fixed the issue.  Can anyone else confirm?

Aeana, 2018-02-15:
I set the speed to normal for my tests above. Some combination of normal speed + formatting with SD Card Formatter seems to make the issue less likely to crop up, but it still crops up eventually.

scotthammack, 2018-02-15:
I'm using a 32GB Kingston SDHC card (model SD-K32G), with speed set to normal. My test was on Super Mario All-Stars running from the SD card. Methodology: load Super Mario All-Stars, start a new game in an empty slot on Mario 1, immediately pause and choose "save and quit," bring up the SD card menu and load into another ROM, then load back into Mario All-Stars and confirm that the save file is present.

I tested this on all four slots and they were all successful; the save files were present as expected. So far, I have not encountered any instances of the save not working properly on my setup. When I get a chance, I'll try some of the other games mentioned above and also try on different speeds.

cameronkeys, 2018-02-15:
I reached the first save room on Super Metroid. Saved the game. Entered the file menu, saved the settings for good measure. Powered off. On startup of the game there was no save present.

I have a SanDisk 16gb class-10 SDHC

jan-kleks, 2018-02-15:
@SmokeMonsterPacks Am I right that when reporting this people might want to provide the exact model and parameters of their SD Card?

leonquest, 2018-02-15:
I dont have a super nt yet, but I imagine you could be experiencing these issues because you are not holding the reset button after powering off. It works the same way in certain everdrives that do not have a battery, in order for the sram to be written to sd memory, hold the reset button when you are done playing.

worleydl, 2018-02-16:
Would love to see code around this but I guess that's not happening. One idea I had was having a dirty bit for the SRAM and automatically writing it out(after a timeout) when it gets flipped?

whitezombie2000, 2018-02-16:
I am having the same exact issue as Aeana when testing this jailbreak firmware. Very erratic with the SRAM, sometimes it saves, sometimes I restart the ROM and the saves are gone.

jr0n, 2018-02-16:
Save failure in Donkey Kong Country:
SmokeMonster hash matching ROM
Windows 10 FAT32 formatted SanDisk Extreme 3/10 16 GB
First save point success. Second save point failure. Menu before hard power off on both saves.

thomas83lin, 2018-02-16:
I can confirm the sram saving bug, played and beat dkc 1 and it only registered the first save point. I saved at every save point before beating the game and it only registered the very first save point. This got me thinking about a feature request I'd like to make. A menu selection to manually save sram either from rom or cartridge. This way I don't have to rely on a automatic function that I can't see if its working or not.

craig588, 2018-02-16:
I played through my NTSC-U and NTSC-J carts of Link to the Past to 100% flawlessly. When I loaded a bunch of Link to the Past Randomizer hacks what seemed to cause them to lose their save was switching between games too quickly. Tried fastest and slowest SD card speeds and the slowest seemed to make them more prone to loosing saves. This is with a relatively old 2GB SD card because I just don't have that many SNES games.

My idea, based on this testing, is maybe adding a little star in the UI when the save data isn't written to the SD card. No idea if that's actually the bug, just throwing something out there.

scotthammack, 2018-02-16:
I tried the Dragon Quest V and VI repro steps from Aeana's original report and was finally able to reproduce this issue. Note this was using the English translation patched versions from the SmokeMonster pack.

With speed set to normal:

- Loaded Dragon Quest V. Started a new game in slot 1. Progressed past the name entry screen.
- Went to the menu and started Dragon Quest VI. Started a new game in slot 1. Progressed past the name entry screen.
- Went to the menu and switched back to Dragon Quest V. My save file was present and started with no problem.
- Went to the menu and switched back to Dragon Quest VI. Pressed a button to skip the intro screens. Before the menu came up, it played some scary music and displayed a screen with the text "6001N->". (I presume this indicates save file corruption.) Continuing to the menu, my only option was to begin a new quest; no saves were present.
- Just to check, went back to Dragon Quest V. This save still worked normally.

I also lost a save on Donkey Kong Country (USA) (Rev 2).
- Started a new game and progressed to the first save point.
- Went to the menu and switched to Dragon Quest V. Loaded my save from earlier.
- Went to the menu and switched back to Donkey Kong Country. All save slots were empty. (I can see through the file browser that a save file was created, though.)

SD card: 32GB Kingston model SD-K32G. Layout:
/SNES
/SAVES
No SD2SNES folders.

iancourt, 2018-02-16:
@thomas83lin: Interesting idea. I'd like an option to redirect loading/saving from the SRAM of an inserted cart to the SD card, basically so the SNT loads/saves from the SD rather than the cart; also the ability to copy saves between the two.

@SmokeMonsterPacks, @frederic-mahe: Some of the feature requests here seem as though they should be directed to Analogue for inclusion in the OFW. Is is still appropriate to raise them here? I know I have quite a few ideas!

phreeze83, 2018-02-16:
dumb question but how can my rom load the savestate ?? any key combination ?

SmokeMonsterPacks, 2018-02-16:
@iancourt I would rather have us help crush bugs here for now versus feature-creeping if possible. That will keep things focused here.

sreuter, 2018-02-17:
@phreeze83 This is about SRAM savegames, not save states :)

jwayniac, 2018-02-17:
I notice that these saves are still contained in the save folder, they're just not being recognized by the rom during the boot.  Played Chrono Trigger for a few hours last night and came back to it later that night to see the save file in the folder but when booted up, it goes straight to New Game.

scotthammack, 2018-02-17:
I got some interesting behavior on ActRaiser (USA). I played through the first action stage and saved during the first simulation stage. Then I went about my business, played some other games, powered down, and came back later. When I loaded the game again, the title screen had a "continue" option as if it found a save. But when I selected it, what I got was some strange broken state -- it seemed to have retained the fast message speed setting I had set on my save, but it made me enter my name again and acted like there was no progress on the save. Then when I selected the "fight monsters" option from the menu, it said I wasn't at a high enough level yet, despite that being the first stage. So there was nothing I could do to progress.

I guess this was just a corrupted save, but it's interesting that it was able to sort of load it but put the game into a broken state. Not sure if that illustrates anything about what might be happening. I tried to reproduce this but couldn't -- after starting a new game and saving again, booting the game now always just acts as if there is no save at all. (But again, there is a .SAV file present in the SAVES folder.)

neonstreetlight, 2018-02-17:
I’ve found this same issue so far with Kirby’s Dream Course. I tried changing the speed of the card but it didn’t solve it. My first thought was that I needed to change my folder structure but now I think this might justify be a bug that needs to be fixed in an update. I’ll keep you posted if I find anything that will help troubleshoot further.

cacophony1, 2018-02-18:
Kosmic Stardust posted a video that demonstrates the save bug here: https://www.youtube.com/watch?v=LUZ2d0yWKcg 

It's a long video but over the course of it you see him:
1. Beat a couple levels in the first world of SMB and then save
2. Verify that the save is still there after a hard reset
3. Get further in the game (beat the first castle) and save again
4. Do a soft reset and verify that the new save is there
5. Do a hard reset and then verify that the game has reverted back to the first save point

This same video also demonstrates the bug about rom loads not reflecting your currently saved settings.

gammy, 2018-02-18:
I tried reproducing thomas83lin's save issue on DKC1. A soft reset left the save intact, while a power cycling the NT broke the save.
The card I tested this with is a 4GB Transcend "Category 4"  SDHC card, which was formatted with a single FAT32 partition, using gparted 0.30.0 (libgparted 3.2).
The SD card options on the NT were left unchanged from factory defaults.

lock14, 2018-02-18:
I believe issues #1 and #2 are related.

My super nt did not remember my scanline and other video settings upon bootup. Everytime I've started playing a game from the SD card I would simply go and adjust the setting back to how they were. But then I thought perhaps this was interfering with the saves somehow.

So to test this, I booted up the console and loaded up 

Donkey Kong Country 2 - Diddy's Kong Quest (USA) (En,Fr) (Rev 1).sfc
69d0f91bfd05837b44023e33a6699aa28fca19cb	d323e6bb4ccc85fd7b416f58350bc1a2	4e2d90f4

and did NOT change the video settings and played to the first save. Then I loaded up 

Super Mario World (USA).sfc
b47bb75d16514b6a476aa0c73a683a2a4c18765	cdd3c8c37322978ca8669b34bc89c804	b19ed489

and played until it's first save point. Then I went back to DKC2 and the save was still there! I beat another level and saved again. Then went back to SMW then back DKC2. Second save was there as well.

I then did the same test but changed the video settings before playing each game and the saves did not stick.

scotthammack, 2018-02-18:
To follow up on my comment earlier about Super Mario World: I have since reproduced this issue in that game too, but it's fairly inconsistent. After my saves worked correctly for the Yellow Switch Palace and the first castle, I proceeded through Donut Plains, cleared the Green Switch Palace and saved there, then cleared the Donut Ghost House's secret exit to unlock the Top Secret Area and saved at that point. I stopped there, returned to the menu, switched games, etc., powered off, and came back later.

When I returned, my save was at the correct point. I then cleared the Donut Ghost House's normal exit, saved, continued to the castle, cleared it, and saved again. After powering off and loading the game again a few minutes later, my save was at the point where the Donut Ghost House had been cleared (both exits), but the castle had not. This would mean that, on my previous play session, the first time I saved worked correctly, but the second did not. This seems consistent with what others have reported, even though I got different results from them earlier.

I cleared the castle again, saved, and did a soft reset using the controller hotkey. My save slot correctly displayed the number of cleared exits as 15, and I was able to load it and start past the castle. I then switched to a different ROM and back, and now the number of cleared exits was 12 (the same as it was at the start of my previous session, with both ghost house exits cleared but nothing after that). After loading this save, beating the castle yet again, saving, returning the menu, then switching to a different ROM and back, I finally got the save to register that the castle had been beaten and display the correct 15 number.

I'm not sure whether this is somehow dependent on the order of the steps taken, or if it just randomly fails some of the time, or what, so I'm just trying to include as much detail as possible. So far, the hypothesis that only the first save attempt after loading a ROM is successful seems mostly, but not entirely, consistent with my experience. It's possible that the inconsistencies are down to user error on my part.

scotthammack, 2018-02-18:
Here's an easy test that breaks consistently (at least for me):
- Start SimCity (USA)
- Start New City
- Use default map, enter whatever for the name
- As soon as the game starts, save (from the load/save menu, fifth icon from the left in the top menu bar)
- Down-select to bring up Super NT menu/file browser
- Switch ROMs and back
- On the main menu, if there's a save, there should be an option to "resume saved city." But it won't be there.

In this game I have never been able to successfully save and reload a game. (For what it's worth I didn't change any Super NT settings while performing this test; only used the file browser to load between ROMs.)

cacophony1, 2018-02-19:
I did some more playing with saves and got the following:

- Played through a couple levels of Super Mario World and saved at the switch palace. Verified that the save worked by switching roms.

- Played through the end of that world (castle) and saved again. The second save seemed to work correctly too (I think I actually power cycled the console but at a minimum I switched roms)

- But later when I returned to SMW again there were zero saves reflected. If I navigate to the saves directory on the SD card there is an entry for SMW though, so maybe the save file got corrupted or something. 

FYI, I'm using this SD card: https://goo.gl/ue8PL8
and I formatted it FAT32 using the built in Windows 7 format operation

scotthammack, 2018-02-24:
I had been chipping away at a save on Super Mario World and found that generally, it would only save correctly the first time I accessed the Super NT menu after starting the game (regardless of how many times I'd used the in-game save system). So I would simply keep playing without using the menu until I reached a save point, save in-game, then go to the menu to write the save to the SD card, and switch games. When I came back, my save would be where I expected.

That worked for a while, but for whatever reason, the latest attempt failed and deleted the save entirely. When this happened, my most recent good save was after Ludwig's castle, at the start of the Forest of Illusion. During this single session, I cleared several levels in the Forest of Illusion, including the Ghost House, the Fortress, using the in-game save system to save after each of those, but not accessing the Super Nt menu. After completing the Switch Palace, I saved, then went to the menu and loaded Super Smash TV. After playing that for a few minutes, I loaded back into Super Mario World, and now all save slots were empty.

GrannySursrise, 2018-02-25:
I am also having issues with save games and some video options not saving. I have been playing around with Zelda, copy and erase, and have a high failure rate. Sometimes it will even revert back to previous saves before erasing everything.

chipperkwah, 2018-02-25:
I'm in the same boat. Sometimes the save will stick, sometimes it won't. Played a little too far into Tactics Ogre.

SasinHyeonmu, 2018-02-25:
With both Final Fantasy II (U) & the j2e Final Fantasy IV (J) translation, I've seen both unable to ever save their games successfully to the SD card.  No amount of time of open up the Super Nt menu appears to flush the save to the card & this isn't experienced with games like Mario All-Stars + Super Mario World.  I've inspected the saves on the SD card after saving in FF IV, opening up the Super Nt menu, and a few minutes later switching to Mario All-Stars, and it appears to be a completely failure to save to the SD card as the save file's md5 hash is identical to before the save attempt and never seems to change (unlike Mario All-Star's)

jtgoen, 2018-02-25:
Have confirmed save data corruption when swapping between Dragon Quest V (J) English translation and Tales of Phantasia (J) English translation.

Steps:
1. Start a new file in Dragon Quest V in the same way as @scotthammack mentioned above
2. Swap over to Tales of Phantasia and progress to where you have control, then leave the town and save.
3. Swap over to Dragon Quest V. An angry sound will happen on boot and will state that your save data is lost forever (must be Dragon Quest V's version of the 6001-> message from DQ VI)
4. Swap back to Tales of Phantasia and attempt to continue the game. All saves will be empty, and attempting to exit this menu will soft lock the game.

xanfir, 2018-03-04:
My save issues seem to be resolved. I have my 16gb Sandisk that I use in my Super Everdrive working. It is keeping saves while switching games in the menu. I have repeated saving, entering new SRAM games and saving, going back to previous games, etc etc. With the last JB firmware, my saves had deleted by now.

neonstreetlight, 2018-03-04:
Just did some testing with 6.5 and this seems to be resolved. SMW saved, loaded Super Punch-Out, saved there, and was able to move back and forth without losing my data. Also tried powering off the console with success. Would be nice to have an option to automatically save SRAM instead of needing to do it each time manually. This would be akin to the Nt Mini feature.

SmokeMonsterPacks, 2018-03-04:
I'm going to close this since most, if not all, save problems have been addressed in fw update v6.5. If new save issues emerge, we'll want a clean slate without all of the pre-6.5 bugs listed.

closed 2018-03-04 by SmokeMonsterPacks