SmokeMonsterPacks/Super-NT-Jailbreak #5: Add the warning that SD cards shouldn't be formatted with OS tools and that one should only use ROMs that have identical checksums to those from good ROM databases

jan-kleks, 2018-02-14:
Read very carefully this thread at Arduino forums, and you will learn why formatting the SD cards with OS tools is definitely not a good idea: [Don't Format SD cards with OS utilities!](https://forum.arduino.cc/index.php?topic=228201.0)

SmokeMonster himself should know how important are good ROM dumps, so I don't need to write anything here.

Let's hope that these two little tips minimize the number of invalid bug reports.

Regards

iancourt, 2018-02-15:
Seems outside the scope of the project to me. Should be up to the user to check that the SD card is correctly formatted and ROM dumps are not free of corruption. I'd personally find such a warning annoying and condescending.

iancourt, 2018-02-15:
Typo: The above should of course read "ROM dumps are free of corruption".

jan-kleks, 2018-02-15:
@iancourt There's nothing condescending about simple tips in the README that can help people IMHO. Many users don't know SD Card Formatter is the ONLY (AFAIK) proper way of formatting SD cards (check the link to the Arduino forums I have posted in my first post), and good dumps are actually quite a serious issue -- look how many preservation groups are there working to create best preservation formats for games. Regards

@SmokeMonsterPacks Okay, so I think these instructions for formatting SD cards could look like this (to cover different circumstances): 

1) Use SD Card Formatter with the "Overwrite" format setting.

2) If you've got an SDXC card (over 32 GB), or a 32 GB card that is causing issues, format it first as 1) and then format it again with [fat32format GUI](http://www.ridgecrop.demon.co.uk/index.htm?guiformat.htm)

3) Linux users would have to find a computer with Windows or macOS [I wasn't able (and I would hope that I'm wrong) to find any tools like SD Card Formatter for Linux, and according to [this](https://forum.arduino.cc/index.php?topic=228201.msg1928855#msg1928855), Linux can't format SD cards properly]

jan-kleks, 2018-02-15:
Sorry about editing the title too many times, but I'm not a native speaker, and sometimes I'm not sure if  I was able to explain everything well...

SmokeMonsterPacks, 2018-02-15:
I've added a line in the README about formatting. I would love to live in a world where everyone verfied their ROMs, but it's probably beyond the capacity of average users.

closed 2018-02-15 by SmokeMonsterPacks

iancourt, 2018-02-15:
jan-kleks, you make a good point, and I take back what I said. You are correct. :)

jan-kleks, 2018-02-16:
@iancourt Thank you, I just like helping people (then I get this nice warm feeling). :)

@frederic-mahe Yeah, this is exactly what I was referring to when I mentioned "proper ROM dumps". Nice!

@SmokeMonsterPacks Adding this fragment about fat32format was a good idea. However, reading carefully through this Arduino thread once again I am more and more convinced that official SD Memory Card Formatter is always the better option. The problem is that the SD Association (of course, consisting of SD card makers) keep the full SD card specification to themselves. The guy who started this Arduino thread knows this stuff because he has exclusive access to documentation (he's also working on the SD card support for Arduino), and this is what he says about formatting SD card larger than 32GB to FAT32:

> SD cards are designed to be formatted with a layout that matches the card's flash chips.  SDFormatter produces the optimal layout by inserting hidden sectors to adjust the layout to chip boundaries.  SDFormatter chooses an appropriate cluster size.  There are no options for the standard format. 

> SDFormatter uses FAT12 for very small cards.  FAT16 is used for cards <= 2GB, FAT32 for cards <= 32 GB, and exFAT for cards larger than 32GB.

**What this means basically is that forcefully formatting the SD cards larger than 32GB and <= 2GB to FAT32 will not be optimal (even with fat32format).** It looks like these cards might have been simply tailored to exFAT. So, theoretically, the best solution would be to support exFAT for the cards larger that 32 GB and to format them with official SD Memory Card Formatter, unfortunately, [we know this is problematic in the case of Super Nt](http://atariage.com/forums/topic/242970-fpga-based-videogame-system/page-200#entry3918167), as well as to tell people not to use the cards <= 2GB at all (but this tip depends on the FAT16 support in the Super Nt, so you may want to ask about this).

The second best solution would be, but I think still more acceptable than the first one, but it's not perfect either, to tell people to use the SD cards that are larger than 2GB but not larger than 32GB with official SD Memory Card Formatter. And if someone would really have to use the card <= 2GB (again, depends on the support of FAT16 in the Super Nt) or larger than 32GB to format them with fat32format, but to warn them it might not work optimally.   

Actually, I strongly suggest everyone who considers formatting SD cards to read the whole Arduino thread, it's really very interesting.

You might think that I'm giving this too much thought, but if you check, for instance, flashcart forums you'll notice that many issues are either due to invalid ROMs or formatting errors. It would be good IMHO opinion to minimize these problems (and invalid bug reports which is really crucial for a project like this) as much as it is only possible with official SD Memory Card Formatter and the use of valid ROMs.

So what do you think on this (after reading everything carefully) @SmokeMonsterPacks and @frederic-mahe?

BTW, I've based this tip from my third post here "If you've got an SDXC card (over 32 GB) format it first as in 1. and then format it again with fat32format GUI" upon the RetroRGB article you linked to @SmokeMonsterPacks, but I got a reply from the creator of fat32format that it's enough to format just with fat32format (but it might be not optimal as I have explained above).

Regards