SmokeMonsterPacks/Super-NT-Jailbreak #6: "Select" stops working to leave file browser.

Llorean, 2018-02-14:
If you're playing a game and press the "return to menu" hotkey, it takes you back to the file browser. From there, if you press "Select" you can return to the game. If you back out all the way to the normal Super NT menus, then re-enter the file browser, you're no longer able to use "Select" to get back to the game. The only way to get back to the game at this point is to go into the normal Super NT menus first, then dismiss them as you normally would, but this doesn't work in the file browser.

scotthammack, 2018-02-18:
Some extra weird behavior related to this bug: you can get it into a real weird state where button inputs are going through to the game when they should be getting captured by the menus, and vice versa.

I made this happen with these steps:
- Boot system.
- Select "Browse SD card."
- Launch in a game (in my case Donkey Kong Country, but I doubt it matters which one).
- Return to menus using the hotkey.
- Back out of file browser to the main Super NT menus using the cancel button.
- Select "Browse SD card" again.
- Press select. Note that it doesn't close the menu.
- Press start to go back to the main menu.
- Close the menu using the hotkey.
- Press B.

For some reason, the file browser will come back up. At this point, you can navigate the file browser and play the game at the same time -- the inputs go to both. If you back out to the main menu and close it again, it will continue to come back every time you press the B button in-game.

(Note: if it makes any difference, I have my confirm/cancel buttons remapped to A and B, respectively, instead of the default Y and B.)

gammy, 2018-02-18:
I believe I just experienced this bug as well. I wasn't recording at the time so I can only describe from memory: 
I was testing out DKC1 (from SD), and, on the first level in the game, entered the NT menu. 
I navigated to hardware->video->scanlines, toggled the extra gain option, and then attempted to return to the game via the select+down hotkey. 
Instead of being thrown back into the game, I appeared to be both in the menu and in the game simultaneously. I experimented for a bit with the menu enter/leave hotkeys, and somehow I managed to get into a normal state again.

I tried to reproduce it, but failed to do so. I probably could, but didn't spend much time on it.

Llorean, 2018-03-04:
Still an issue in the new update.

SmokeMonsterPacks, 2018-03-04:
Please list the games where this is still occurring in fw v6.5

Llorean, 2018-03-04:
Still occurring? The changelog doesn't indicate this was fixed, I was mostly just documenting that none of the unrelated changes have changed this issue's status.

This is not the "fixes glitch with some games disabling the "return to menu" shortcut," issue. Rather this is an issue when you're already *in* the menu, and trying to return from it, to the game. The "return to menu" issue was one where you could be in a game and unable to use the shortcut to get out of the game, into the SNT menus. They're basically mirror images of each other, really, but the issue that was fixed apparently has something to do with games reading controllers

As far as I can tell, this issue is game independent. I mean I've seen it in Super Metroid, Final Fantasy 6, Front Mission: Gun Hazard, and others.

scotthammack, 2018-03-04:
I can confirm that the repro steps I listed above still produce the same behavior in Donkey Kong Country. I also tried the same steps in Super Mario All-Stars and got the same results. I agree this is probably game-independent.

SmokeMonsterPacks, 2018-03-05:
Ahh, I was reading through to many issues at once. Correct, this is a general bug.

Llorean, 2018-03-05:
Figured there was a non-zero chance of this accidentally getting closed because of the similarity with the other issue in the patch notes, which is why I followed up to say "Hey, it's still here!"

Honestly, this is the biggest issue for me in using the jailbreak, because if I'm not careful to leave the file browser exactly the right way, things end up somewhat unusable until I reboot the system.

cacophony1, 2018-03-05:
I can confirm that this issue is 100% reproducible. Not that I think the game matters, but I tried Super Mario All Stars and Final Fantasy III. Once I had used CANCEL/BACK to navigate from the file browser to the main menu I could never again use SELECT to return to the game from the file browser (it only works from the menu areas outside of the file browser)

frederic-mahe, 2018-03-05:
Someone suggested the following workaround:
> if the file menu is exited, it becomes impossible to re-enter the game via select or save the RAM. Use start if you wish to enter the settings menu while in the file menu. This is safe and will allow re-entering the game via select and saving RAM. RAM will NOT be saved if the setting menu is exited via select and re-entered with the hotkey- only doing this in the file menu will work.

@Llorean would that make the jailbreak more usable for you?

Llorean, 2018-03-05:
Not really. It's possible to avoid the issue just by never entering the menu in the first place, just going the file browser then back to the game. But workarounds aren't a solution for if you do happen to hit the wrong button just once, and lose an arbitrary amount of progress.

And given that "B to go Back" is the right thing to do in literally every screen but exactly one page of the File Browser, and if you do it on that page, things start breaking, it's an issue a lot of people are going to run into just because of the way you're trained to use the UI in general. A "quick" fix would just be to prevent B from taking you out of the file browser and back into the main menu, since then you'd have to use Start. A more complex fix would be to figure out why doing this messes things up, and resolving that so that B continues to work as expected. Either way, it just seems like a pretty significant issue since a stray button press can break saving until you power cycle.

cacophony1, 2018-03-05:
@Llorean @frederic-mahe In this case the suggestion is to use the "start" key to get from browse roms to settings. This is the developers intended path and works fine with no issue. But the point of this bug is that if the user uses the alternative path of "back/canccel" to get to settings then it completely breaks the ability to use "select" to bring down the menu in certain areas.

Llorean, 2018-03-05:
We know what the suggesting is, it doesn't change that there's a bug in need of fixing, and one with serious consequences.

cacophony1, 2018-03-05:
Yeah, I was mainly just trying to clarify a misleading statement you made above where you referred to the proposed solution as a workaround, when in fact it is the developer's intended path.

scotthammack, 2018-03-05:
Even if you only use the intended path, it doesn't completely solve the issue, as there's still no way to access the codes menu (or the included Turrican games) from the file browser without losing the ability to save.

frederic-mahe, 2018-03-06:
@Llorean @axebox @scotthammack @gammy @cacophony1 I think this issue and the issue #48 could be renamed. For example:
> leaving the file browser prevents from saving the active game

Would you help me improve on that?

Llorean, 2018-03-06:
Something like "Backing out of SD Browser to Super NT Menus while in-game breaks multiple features?"

frederic-mahe, 2018-03-06:
@Llorean I would like to have a slightly different title per issue, one could mention the select issue and the other the save issue. What about?
> Backing out of SD browser to menu while in-game breaks save function

> Backing out of SD browser to menu while in-game breaks functionality of select button

Llorean, 2018-03-06:
That seems fine. Though there's also the related issue where the file browser passes button presses on to gameplay.

From my perspective, the file browser is just breaking in a variety of ways all at once. There's at least three unique symptoms (can't leave with select, can't summon save prompt, controls pass on to the game) that all occur from the same action (leaving with the B button).

I see it more as "B button in the root of SD card breaks SD card browser functionality in many ways" than a host of individual bugs that all happen to trigger with the same single button press.

cacophony1, 2018-03-06:
@frederic-mahe The two versions you suggested sound fine, though I would make a small adjustment:

Using BACK to exit SD browser while in-game breaks save function
Using BACK to exit SD browser while in-game breaks functionality of select button

frederic-mahe, 2018-03-06:
@Llorean you are probably right, the three issues are likely related but I'll keep them separated for now. @cacophony1 thanks for the edits.

neonstreetlight, 2018-03-06:
Just experienced this bug. Since I could not use the select button, I could not trigger the prompt to save SRAM. Lost all progress since my last save in Final Fantasy V unfortunately.

TheYokai, 2018-03-08:
I posted this in #48 but it is worth mentioning it here as well:

There seems to be a lot of state issues here -- I think the whole menu system needs to be simplified. Let's enumerate problems:

If you return to the main menu while playing a game loaded from SD card, when you return to the SD Card menu not only can you not save anymore but you also can't hit select to return to the game while in that menu.

There are two settings screens (one in the main menu and another one in the SD card menu by pressing start) and "Run Cartridge" is both in the main menu and in the SD card menu where it doesn't seem very necessary.

The main menu should really be the central hub from which you can access every other menu (Run Cartridge, SD Card, Settings, Tools, etc). and going there shouldn't break anything. Linking from one menu to a totally different menu just to prevent people from returning to the main menu is just a work around for what is really a flaw in the current design. If the user could go back to the main menu without breaking anything then you wouldn't need run cart or settings shortcuts within the SD card menu at all.

I do think it would be handy to have the ability to manually dump SRAM to SDCard from the SDCard menu when a user might accidentally dismiss the save pop-up or when state bugs like these occur.

SmokeMonsterPacks, 2018-03-14:
I've added this to "Known Issues" in the [README](https://github.com/SmokeMonsterPacks/Super-NT-Jailbreak/blob/master/README.md).

SmokeMonsterPacks, 2018-03-27:
This should be resolved in the new JB v6.5 firmware.

closed 2018-03-27 by SmokeMonsterPacks