SmokeMonsterPacks/Super-NT-Jailbreak #7: Does not install

nedelman, 2018-02-14:
I downloaded the .bin file, and saved it to the root directory of a freshly formatted 32GB SD card. I formatted the card using the "SD Card Formatter" application from www.sdcard.org.

After inserting the SD card into the Super NT and powering the system on, the power LED flashes red with an apparent pattern. The pattern seems to be "4 flashes, 2 flashes", which then repeats over and over.

jan-kleks, 2018-02-14:
What type of an SD card is this?

nedelman, 2018-02-14:
Samsung EVO micro SD in a full SD adapter. The exact same brand/type I'm using in my NT Mini.

jan-kleks, 2018-02-14:
@nedelman Can [this](http://atariage.com/forums/topic/242970-fpga-based-videogame-system/?p=3962714) be the problem?

nedelman, 2018-02-14:
I'll try again and see what happens if I just leave it for a longer amount of time.

jan-kleks, 2018-02-14:
@nedelman Just don't forget to report back.

SmokeMonsterPacks, 2018-02-14:
Make sure to give it a few minutes to flash. I've just added a note about this in the README.

nedelman, 2018-02-14:
I left it for about 15 minutes and there was no change. The flashing pattern was the same as before.

And just to be more explicity about my SD card's setup:
-The card is formatted as FAT32 (as instructed)
-The only file in the card's root directory is the .bin file (as instructed)
-The only directory on the card is "SNES"

SmokeMonsterPacks, 2018-02-14:
Could you please try formatting the card following these instructions? http://www.retrorgb.com/romcarts.html#format

If that works, I can point people there for a solution.

jan-kleks, 2018-02-14:
@nedelman If the @SmokeMonsterPacks' solution does not work you can also try to use the Overwrite format option of SD Card Formatter.

nedelman, 2018-02-14:
I figured out the problem, and it's a bit dumb, but may help others in the future:

Basically, within Chrome I was right clicking on the .bin file on the Github page and selecting "Save Link As..." which is actually saving the page, not the .bin file. After going to the page and then correctly saving the .bin file, everything seems to be working.

Also I checked out the retrorgb.com link, and the "SD Card Formatter" screenshot is outdated. The most recent version (5.0) does not look like that or have the options shown in the screenshots.

closed 2018-02-14 by nedelman

nedelman, 2018-02-15:
Also, thanks for the help. I'm really enjoying the jailbreak.