SmokeMonsterPacks/Super-NT-Jailbreak #8: Add checksum for the firmware

jan-kleks, 2018-02-14:
So that people know the file was not corrupted, for instance, during downloading.

SmokeMonsterPacks, 2018-02-15:
Great idea, thanks. I'll add it to the README.

SmokeMonsterPacks, 2018-02-15:
Done, thanks again.

frederic-mahe, 2018-02-15:
In my opinion, that should be solved upstream by creating a MD5SUM file alongside the firmware:
```sh
md5sum snt_firmware_verJB6.4.bin > MD5SUM
```
If this is done before the first network transfer, then we can be pretty sure that the file is not corrupted.

frederic-mahe, 2018-02-15:
I've just added a _post-facto_ MD5SUM, but that should be done upstream.

SmokeMonsterPacks, 2018-02-18:
I've put in a request upstream for checksums to be included with firmwares going forward.

SmokeMonsterPacks, 2018-03-04:
No follow up. Issue marked as resolved.

closed 2018-03-04 by SmokeMonsterPacks

SmokeMonsterPacks, 2018-03-06:
Upstream MD5 checksums added for both releases.