SmokeMonsterPacks/Super-NT-Jailbreak #10: Graphical Glitches in Power Rangers Fighting Edition

Moonbase1, 2018-02-15:
Power Rangers Fighting Edition shows graphical glitches randomly during game play - mostly commonly during the character select screen, and game over screen. Sometimes when a fight loads, the life bars glitch and the game soft locks. Happens with an authentic cartridge as well as a ROM verified good dump.

[image]
[image]

burk3, 2018-02-16:
does this happen on the stock firmware as well?

Moonbase1, 2018-02-16:
Yes it does

Moonbase1, 2018-02-16:
Yes

frederic-mahe, 2018-02-16:
@Moonbase1 , thanks for reporting that bug! Just to be sure we are talking about the same cartridge/ROM, could you please confirm the information below?
```
Mighty Morphin Power Rangers - The Fighting Edition (SNS-A3RE-USA)

ROM's SHA256: 1b85c0690aa156a255c7f79e133e453345452698fa98abf8df744c262d0cf865
ROM'S SHA1:   80a856dba0f12fd1e9eded75d219b854910e990f
ROM's MD5:    bdfcd02f353bc9745dda4556fb7db08a	
ROM's CRC:    460b0a60
```
(confirming one of the checksum values is enough)

Moonbase1, 2018-02-16:
Yes that's the same ROM.

burk3, 2018-02-16:
@frederic-mahe I'm curious, is it your intention to accept bugs for the official firmware?

frederic-mahe, 2018-02-16:
@burk3 bugs affecting the official firmware should be reported to Analogue. This GitHub repository is dedicated to the jailbreak firmware.

That being said, it is likely that bugs affecting the jailbreak firmware will also affect the official firmware, especially graphical glitches or cartridge/ROM specific issues. We report those bugs to Analogue, so they can be corrected in the official firmware too (and eventually in the software emulator higan), for the benefit of the whole community.

eabsolidcore, 2018-02-18:
In other words they are watching and will fix . Found bugs in prince of Persia and super James pond lag .

SmokeMonsterPacks, 2018-02-25:
Please re-test this issue with the new official firmware v4.3 using your original cartridge. 

https://support.analogue.co/hc/en-us/articles/360000557452-Super-Nt-Firmware-Update-v4-3

frederic-mahe, 2018-03-04:
@Moonbase1 please re-test this issue with the new jailbreak firmware.

SmokeMonsterPacks, 2018-03-04:
No follow up. Issue marked as resolved.

closed 2018-03-04 by SmokeMonsterPacks

scotthammack, 2018-03-04:
I played through three matches using the ROM on JB6.5 and didn't encounter any graphical glitches or soft locks.

Moonbase1, 2018-03-04:
It takes more than 3 matches sometimes - I will retest tonight

SmokeMonsterPacks, 2018-03-04:
Thanks for the follow up, Scott!

Rahzadan, 2018-03-24:
Just tested this for about 30 minutes on both Super Nt 4.4 and Real SNES hardware using SD2SNES to make comparisons. Did not encounter any of the suggested glitches or issues.