SmokeMonsterPacks/Super-NT-Jailbreak #11: Scanline and aspect ratio settings not retained after powering system off/on

poorstargazer, 2018-02-15:
problem was exhibited with video settings at 1080p60 and 720p60

set resolution and scanlines to normal then saved settings
Power system off/on and load game from SD to see there are no scanlines displayed
Bring up the menu in game and check setting, normal scanlines are selected. I can either toggle off/on or just confirm the currently selected option and then scanlines appear and the aspect ratio changes slightly.

rgdelato, 2018-02-16:
Is this the same issue as #1?

poorstargazer, 2018-02-16:
After looking over all the posts I would say it seems to be all related. The video settings bug is seen immediately when powered on and after corrected I haven't seen it come back until I power cycle the system though.

SmokeMonsterPacks, 2018-02-16:
I'm going to close this since it is already being discussed in #1

closed 2018-02-16 by SmokeMonsterPacks