SmokeMonsterPacks/Super-NT-Jailbreak #12: Creating an issue template to facilitate valid bug reporting

jan-kleks, 2018-02-16:
[Creating an issue template for this repository](https://help.github.com/articles/creating-an-issue-template-for-your-repository/).

In the case of your project this could contain questions such as "whether the game was tested with original hardware as well?", "what kind of ROM was used (provide the checksum)?", the questions about most important Super Nt settings, etc.

Consult this with the dev behind the jailbroken firmware to see what other question would be useful.

frederic-mahe, 2018-02-18:
There is now an issue template file automatically loaded when opening a new issue. Suggestions on how to improve it are welcomed.

SmokeMonsterPacks, 2018-02-21:
Well done, Frederic. It looks like the new form is really working well.  going to close this to keep things tidy.

closed 2018-02-21 by SmokeMonsterPacks

jan-kleks, 2018-02-25:
@frederic-mahe @SmokeMonsterPacks [The article about testing games with the Super Nt for bugs has appeared on the Nerdly Pleasures blog](https://nerdlypleasures.blogspot.com/2018/02/super-nt-testing-and-capturing.html), maybe it will give you some more ideas on how to expand the template.

SmokeMonsterPacks, 2018-02-26:
That is an excellent article and it's great to know that greathierophant is working on bugs. He will make swift work of them and can provide the technical explanation of why they happen. He helped me burn though hundreds of issues with the NT Mini Pack over the last year.

That said, I think we can keep the template more general in order to get a wider range of bugs. Then more technical users can hash then out following nerdlypleasure's guidelines. I fear that if we get too specific it could isolate people.