SmokeMonsterPacks/Super-NT-Jailbreak #13: 480p screen size too narrow

aarontees, 2018-02-16:
If the resolution is set to 480p, the 4:3 for 16:9 setting renders narrower than 8:7. It also seems that regardless of the screen size setting, the boot animation also renders narrower than it should as well.

aarontees, 2018-02-16:
Seems like the issue occurs with the official firmware as well...

frederic-mahe, 2018-02-16:
If this issue is confirmed, we should keep it opened.

aarontees, 2018-02-16:
When I went back to using the official firmware, I still had the narrow boot screen in 480p. When I first got the NT,  I thought I remembered the boot screen being the correct ratio in 480p. I don't know anymore because regardless of whether I use the jailbroken firmware or not, I have a narrow boot screen in 480p with all screen sizes, and both the 4:3 screen sizes are far narrower than 4:3. My TV is 720, and the picture mode isn't set to stretch.

Galgomite, 2018-02-17:
@aarontees The Super NT is always outputting a 16:9 signal, I think. So your TV needs to be in stretch/ 16:9 mode to display the 4:3 picture properly.

aarontees, 2018-02-17:
Oh interesting, I toyed with that, but then the menu UI is stretched, and 4:3 for 16:9 looks really grainy. It almost seems as if there should be a separate 4:3 picture size for 480p.

SmokeMonsterPacks, 2018-02-18:
So is this issue a TV setting issue? If so it can be closed.

aarontees, 2018-02-18:
Not a TV settings issue. There's no way for me to get a 4:3 picture in 480P with one of the picture size presets. I should't have to use a stretch setting on my TV. If the NT is gonna output 640x480, it should be outputting a 4:3 picture.

closed 2018-03-04 by SmokeMonsterPacks