SmokeMonsterPacks/Super-NT-Jailbreak #14: Unholy Night graphical glitches

Moonbase1, 2018-02-16:
Unholy  Night when in a fight will randomly turn the screen into garbage. Once the fight ends, the graphics return to normal. On official firmware the game hangs before the title screen.

ROM DATA ROM CRC32: DF99025E

Have attempted to play with cartridge and game loads and plays, have not confirmed if the cart glitches as well

frederic-mahe, 2018-02-16:
@Moonbase1 , I don't know this game (name or CRC32 not found). Could you please give more info? (NTSC or PAL? cartridge serial, etc.)

Moonbase1, 2018-02-16:
[image]

Error confirmed with original cartridge

Moonbase1, 2018-02-16:
It is a indie game released this year on cartridge. NTSC. Analogue told me they bee aware of it not running on firmware 3.9, and they were working on it. With this JB firmware it lets you play but graphical glitches.

SmokeMonsterPacks, 2018-02-25:
Please re-test this issue with the new official firmware v4.3 using your original cartridge. 

https://support.analogue.co/hc/en-us/articles/360000557452-Super-Nt-Firmware-Update-v4-3

RevQuixo, 2018-02-25:
I was able to getting halfway through the first fight before the background glitched out.  Then in the 2nd fight my sprite disappeared and the screen started jumping around.   Still broken under 4.3

SmokeMonsterPacks, 2018-03-04:
No follow up. Issue marked as resolved.

Moonbase1, 2018-03-05:
DId not have a chance to retest until now - background still glitched on new JB 6.5 lasted half a round.

SmokeMonsterPacks, 2018-03-05:
Thanks, reopened

SmokeMonsterPacks, 2018-03-18:
Kev has Unholy Night fixed in the next official firmware update.

SmokeMonsterPacks, 2018-03-26:
This has been fixed in v4.4

SmokeMonsterPacks, 2018-03-28:
RevQuixo dumped his European region Unholy Night cart via CopySNES and it crashes after finishing the first match. 
MD5: ffe5445ac39f30efc397899351d12b07

SmokeMonsterPacks, 2018-03-28:
I'm going to open a new issue for this since Unholy Night is now fully working, it just isn't dumping correctly due to faulty ROM programming by the developers.

closed 2018-03-28 by SmokeMonsterPacks