SmokeMonsterPacks/Super-NT-Jailbreak #15: Prepare FAQ

jan-kleks, 2018-02-16:
So that people would not post similar questions over and over again.

I mean things like this: https://github.com/SmokeMonsterPacks/Super-NT-Jailbreak/issues/4

Or like this: https://github.com/SmokeMonsterPacks/Super-NT-Jailbreak/issues/9

Of course, consult this with the dev behind the jailbroken firmware.

jan-kleks, 2018-02-16:
@SmokeMonsterPacks @frederic-mahe Also, you might want to talk with the dev behind the jailbroken firmware about using this GitHub as **the** bug tracker, looking for various bugs through forums and YouTube comments can get really messy at times. And users sometimes even don't know whether their bug has already been reported or not...

SmokeMonsterPacks, 2018-02-16:
I'd rather keep all of the relevant info in the README.

closed 2018-02-16 by SmokeMonsterPacks