SmokeMonsterPacks/Super-NT-Jailbreak #17: Serious Damage messages

theboy181, 2018-02-16:
Some ROMs I have encountered display messages detailing that  the game will either damage the system or that the game pak is not designed for my system. I was under the impression that the Super NT was very friendly with the low voltage carts, and repo's on the market. I'm struggling to understand the issue with ROMs.

[image]
[image]

Llorean, 2018-02-17:
This is more of a copy-protection message. In the case of cartridges, this can mean they're just not clean enough. DKC3 gave me warning about "not working with devices designed to copy cartridges" until I took it out and gave it a serious contact cleaning, and then all was well. With ROMs, it may be an issue of setting the region.

SmokeMonsterPacks, 2018-02-17:
Were you attempting to play Japanese region ROMs while the console was set to US region, or vice versa?

theboy181, 2018-02-17:
I didn't even see any region settings, except PAL and NTSC 
As far as I could tell its set to NTSC, and the ROMS are NTSC.

Moonbase1, 2018-02-17:
I encountered this issue with super punch out on a real SNES. NTSC authentic cart on an SNES Mini. I had to clean the cart. 

As far as low voltage carts, unholy Night doesn’t run properly either. Analogue was made aware of it and working on it in an email they sent me

SmokeMonsterPacks, 2018-02-17:
@theboy181 Please list the filenames of the ROMS that exhibited this message and their hashes so others can try to recreated the error.

theboy181, 2018-02-17:
Any recommendations on program to review the hashes?

frederic-mahe, 2018-02-17:
On Mac and Linux, there are command line tools (`sha1sum`, `md5sum`). On Windows, you can use the [FCIV](https://support.microsoft.com/en-us/help/889768/how-to-compute-the-md5-or-sha-1-cryptographic-hash-values-for-a-file) utility.

theboy181, 2018-02-17:
Algorithm : SHA512
Hash      : B31A6F340596DA409DBF89E04B0537C56349BABF08C8A5A5656FD2D9ADFC0852B206A7B36FD6B7DB37A68AA9C54E569BCE428AD0EF6
            C6B916295C947CDC8FBA5
Path      : L:\SNES\Super Mario All-Stars.smc

Algorithm : SHA512
Hash      : 38F727FE3FA2FCAE5FF3F2F73BF2861ADC766A5319DA3E3FE7A8F5B6CBE4EC674244096EDC82FAD16A978D96F155081624734B7A811
            DB369005B2114D971CB82
Path      : L:\SNES\Donkey Kong Country 2 - Diddy's Kong Quest.smc

Algorithm : SHA512
Hash      : 86EB15537AC3D404421C238CA9CACE80243C0E5B20AB0F62C145B9683F7F0A7B5A888B8E1DBB301C0DE0E9354D82C8B9345F5988BA2
            5FDB28CF737A33F3C69C5
Path      : L:\SNES\Donkey Kong Country.smc

Let me know if you need more.

SmokeMonsterPacks, 2018-02-17:
Those are very questionable quality ROMs, and they have headers. Try seeking out .SFC ROMs and report back. For example, your Super Mario All-Stars ROM is from neither No-Intro or GoodSNES. You probably downloaded them as a package somewhere.

frederic-mahe, 2018-02-17:
@theboy181  Here are the SHA512 values I get for the US versions of the games you've listed:
```
9566975d82397faac75fc4bdfc05ce40a13856b82c329697ec6b99d12972db57964a1531e447d98ccf1f3865341b01c5203802c52da7b5b576fb4576482f49af  Donkey Kong Country 2 - Diddy's Kong Quest (USA) (En,Fr) (Rev 1).sfc
47963ead0093771f87056f33a4aaa8f1680ec333061ed67efc6fa3d02d727f9a13648ba55cdc6dc697eae84b16143eec09fe552d3908c4e5c231e3ff0eb8e9dc  Donkey Kong Country (USA) (Rev 2).sfc
36fbf08991ab27f94e7940947d7764fee88297a3c1854f869de542caded89162c0c202a1ec9c4528e42338b870c1561b01f9c1b30b140275b8c88386fc732ad8  Super Mario All-Stars (USA).sfc
```

theboy181, 2018-02-17:
Based on the Super Mario All-Stars should it not run?

frederic-mahe, 2018-02-17:
What I meant is that I cannot rule out that you are using valid ROMs of the Donkey Kong games, but I am sure that your Super Mario All-Stars ROM is not valid (assuming it is the US version).

theboy181, 2018-02-18:
my mistake, I miss read what you said.

borti4938, 2018-02-20:
Too much wrong information in here. No bad ROMs, no durty contacts!
This message usually appears, when a game checks register $213F (a PPU2 register) and looks at bit 4. Bit 4 gives the information whether the console runs at 50Hz or 60Hz.
Kevtris decided to output at this bit the region which you set in menu rather than simply output the bit value which is expected by the game to continue. In my point of view a bad design idea as you can see here.
With the sd2snes you can circumvent this by enabling "Auto region patching" (or similar) but this only works on 'real' consoles. The SuperNt outputs the register read to the cartridge address bus b but the value written on the databus is not read in this case.

frederic-mahe, 2018-02-20:
@borti4938 thanks for this technical explanation. If I follow you correctly, in this case the Super Nt is too faithful to the way "real" consoles behave. I'll go ahead and mark it as a bug.

SmokeMonsterPacks, 2018-02-21:
@borti4938 Thank you for the explanation. Please keep an eye on the issues and lend us your expertise.

borti4938, 2018-02-21:
Yes, I'll do. Although I'm not that interested in the jailbreak, I'm interested in this specific behaviour. I was disappointed about that since I saw the review from MLiG. Although I contacted Kevtris, I felt like to be alone on that. Maybe with the jailbreak this specific 'problem' gets more in point of interest.

I would not say it's a bug; more like a feature request to bypass this type of region lock.

iancourt, 2018-03-02:
@borti4938, @frederic-mahe, I note that when using an sd2snes cart, the sd2snes configuration reports "SuperCIC detected but not used", just like my SuperCIC-modded original PAL SNES. So does the Super Nt have an implementation of ikari_01's SuperCIC? If so, would an option to enable auto region and auto refresh rate selection be technically possible?

borti4938, 2018-03-03:
With 99% this is technically not possible. The SuperNt is not designed that the region is controlled by the SuperCIC. It’s a reduced version of the SuperCIC-lock with the Pairmode removed!
I really don’t know why they haven’t took the SNESCIC-lock-resync instead which would be a much better choice for the SuperNt.
50/60Hz is not controlled by the SuperCIC anyway.

Also auto region patching by the sd2snes is not possible as explained beforehand.

iancourt, 2018-03-03:
Thank you for clarifying, @borti4938.

SmokeMonsterPacks, 2018-03-04:
This should be addressed in v4.4

borti4938, 2018-03-05:
No, it's definitely not.
It's neither stated in the change log nor it's implemented 'as hidden' feature.

SmokeMonsterPacks, 2018-03-05:
Ok, reopened.

SmokeMonsterPacks, 2018-03-28:
Kev wants to smooth out region issues in a future official fw, so I'm going to close these to get the focus back on bugs.

closed 2018-03-28 by SmokeMonsterPacks