SmokeMonsterPacks/Super-NT-Jailbreak #18: Bakup Cartridge - feature request

theboy181, 2018-02-16:
Would it be possible to backup my personal cartridges to SD card using the Super NT? Maybe even handling a  cartridge save  management system too?  I have a bunch of games that I would love to backup, and with a save backup I could replace my SRAM batteries knowing that I have a good backup of my SRAM on the Super NT.

SmokeMonsterPacks, 2018-02-17:
We have no information on this.

closed 2018-02-17 by SmokeMonsterPacks

theboy181, 2018-02-17:
Was it closed because it can’t be done? Or you didn’t feel it’s a worthy feature request?  :(

romevi, 2018-02-17:
Iirc it took kevtris a long time to make it for the NES, so making it for the SNES would be a different beast than jailbreaking it to play ROMs.

frederic-mahe, 2018-02-17:
@theboy181 This is a worthy feature, but turning the Super Nt into a cartridge dumper is out of the scope of the jailbreak firmware for now.

Such a device exists for the Megadrive ([universal-mega-dumper](https://db-electronics.ca/2018/01/17/universal-mega-dumper-tech-demo-video/)). There is no doubt that would be very interesting to have one for the SNES.

SmokeMonsterPacks, 2018-03-27:
Cartridge dumping is now possible via the CopySNES feature in JB v6.6

theboy181, 2018-04-01:
This was a surprise.. I wonder if I should knock my other ideas around now?