SmokeMonsterPacks/Super-NT-Jailbreak #20: Link to the Past Randomizer hack crashes

craig588, 2018-02-17:
This has happened to me twice now, both times when entering a cave (but not the same cave). Restarting allows me to progress with no issues. I don't have a flash cart to test it to find out if it's an issue with the built in loader or a problem with the rom, but I'm guessing there's something different about the Randomized hacks. The rom size is increased from 1MB to 2MB. No crashes yet from playing dozens of them on BSNES through Bizhawk. I've played through my LTTP NTSC-U and J carts and didn't have that problem. I think my next step could be playing an unaltered original from a SD card if you'd like me to check out if there's any problems with that.

I know that hacks can introduce all new problems because of insufficient testing on the part of the hack author, but I have seen people play complete games without crashing on a SD2SNES using a original console.

SmokeMonsterPacks, 2018-02-17:
Please list the hack version that you're using and a hash of your ROM, so others can test it.

scotthammack, 2018-02-19:
There has been some discussion on the atariage thread recently about being able to [crash the unaltered version of Link to the Past while walking in and out of Link's house](https://atariage.com/forums/topic/242970-fpga-based-videogame-system/page-316#entry3965833). I wonder if this could be related. It has been reproduced on the Super NT using the ROM from the SmokeMonster pack as well as using the cartridge, though it is somewhat intermittent.

craig588, 2018-02-19:
I was able to reproduce this with an orignal NTSC-U rom CRC 777AAC2F. It seems very similar to the hack crash. Could be that my inputs get sloppy after playing a while and I didn't notice switching directions as I was entering areas.

SmokeMonsterPacks, 2018-03-04:
No follow up. Issue marked as resolved.

closed 2018-03-04 by SmokeMonsterPacks