SmokeMonsterPacks/Super-NT-Jailbreak #21: Monopoly Community Chest Hangs

coughlanio, 2018-02-17:
Possible bug/glitch with Monopoly (from the SD2SNES pack, CRC matches), in two player mode, when getting a community chest card, the game appears to hang, with no way to progress.

**Expected Behavior**

Can proceed through a full game of Monopoly without issue, including when getting a community chest card.

**Actual Behavior**

Game hangs after receiving a community chest card.

frederic-mahe, 2018-02-17:
Thanks for reporting this bug @coughlanio 

Could you please confirm the ROM is _Monopoly_ (USA) (Rev 1)?
```
sha1: beddd101739f5e1a6e3abd1542cccd341e6e5263
md5: f478e1eddc87352ed35205d6cfe017ce
crc32: 88d54085
```

The corresponding cartridge would then be _Monopoly_ (`SNS-ML-USA`, revision `SNS-ML-1`). To check whether or not the bug also affects the official firmware, it would be great if someone could check with an official cartridge.

coughlanio, 2018-02-17:
That is indeed the version 👍

Jagasian, 2018-02-19:
I am able to reproduce this issue using official Super NT firmware 4.1 and sd2snes firmware 0.1.7e.  The game also locks up when getting a chance card.

Jagasian, 2018-02-24:
I downgraded to SNT firmware 4.0 and the game no longer locks up when the Chance or Community Chest animation is played, but there is still an issue.  The animation exits back to the board a second after it is started, so it is impossible to see what card you just picked up.

Jagasian, 2018-02-25:
The bug still repros in firmware version 4.3

frederic-mahe, 2018-02-25:
thanks for your feedback @Jagasian

frederic-mahe, 2018-03-04:
That bug should be solved now with the new jailbreak firmware. Please, give it another try @coughlanio

SmokeMonsterPacks, 2018-03-04:
Holdmybeer on Discord confirmed this has been resolved

closed 2018-03-04 by SmokeMonsterPacks