SmokeMonsterPacks/Super-NT-Jailbreak #23: Crash: Home Alone (USA)

scotthammack, 2018-02-17:
- Load Home Alone (USA).sfc
- On title screen, start a new game
- Press start to skip the intro cinematic
- As soon as the game starts, the screen goes black and the game does not respond to input. The music continues playing.

SHA1: e1f96f63603f4d899ec8e29975d0cdf0170cde73
MD5: 87447e71a602b02797054a4d80349c20

Unfortunately I don't have a real cartridge to test against.

frederic-mahe, 2018-02-17:
@scotthammack , thanks for reporting that bug. I've checked against byuu's [dump database](https://preservation.byuu.org/), and that's a valid ROM.

It would be ideal if someone could test the real cartridge on a Super Nt with an official firmware.

Ghettoblaster81, 2018-02-18:
Confirmed working. USA cart on original firmware.
[image]

SmokeMonsterPacks, 2018-02-18:
@scotthammack Could you please test using a different SD card if possible?

scotthammack, 2018-02-18:
Replicated using the same ROM loaded off a 4GB Toshiba MicroSD-HC card. The results were identical to my initial report.

SmokeMonsterPacks, 2018-02-25:
@Ghettoblaster81 Please re-test this issue with the new official firmware v4.3 using your original cartridge. 

https://support.analogue.co/hc/en-us/articles/360000557452-Super-Nt-Firmware-Update-v4-3

Ghettoblaster81, 2018-02-27:
@SmokeMonsterPacks it wasn't me with the error, that was @scotthammack 

I just tested it with the original cart that worked on original 4.0 firmware. I guess it still works on 4.3 firmware. Will update my firmware asap.

frederic-mahe, 2018-03-04:
The new jailbreak firmware should fix that crash. Please, give it another try @scotthammack .

scotthammack, 2018-03-04:
Confirmed fixed on JB6.5 firmware.

closed 2018-03-04 by scotthammack