SmokeMonsterPacks/Super-NT-Jailbreak #24: Glitched graphics in Super Adventure Island (USA)

scotthammack, 2018-02-17:
- Load Super Adventure Island (USA).sfc
- Start the game
- The background graphics for the first level are glitched out.
[image]

SHA1: 74a3671878b1d7c8b3fd5a14677c16b2d5566af0
MD5: 206e6c5c05ac1a83dde21400325a4a7d

frederic-mahe, 2018-02-17:
Hash value matches byuu's dump database.

cacophony1, 2018-02-18:
I do not see this issue with the Super Adventure Island rom that I have (which came out of a recent Everdrive pack off archive.og). But it looks like mine has a different SHA1 and MD5 value.

scotthammack, 2018-02-18:
Oddly, after reading @cacophony1's comment, I tried again and this issue didn't occur. I have no idea why it would have changed. Maybe it just didn't load the ROM correctly when I first got the issue, for whatever reason?

I guess this bug can be closed?

SmokeMonsterPacks, 2018-02-18:
Thanks for the follow-up. It could very well be SD card related, since MSU1 functionality in SD2SNES is very picky about that. If it crops up again, maybe try another fast card.

closed 2018-02-18 by SmokeMonsterPacks