SmokeMonsterPacks/Super-NT-Jailbreak #27: small final fantasy 3 glitch

masterninjah, 2018-02-19:
[image]

garbled backgrounds in final fantasy 3 right after the fight with vargus. it goes back to normal after you go to another screen.  using the original cart, not sure what revision it is.
im using the jailbreak firmware also

i also have the problem with the 4x scale making my screen smaller than what i set it to.  if i turn the scaling off it pops back to normal then i can set it back to 4x

gammy, 2018-02-19:
I'm curious: @masterninjah, does this happen on the stock firmware (as of writing, v4.1) as well?

masterninjah, 2018-02-19:
im not sure, i havent played this on anything but the jailbreak firmware

saftle, 2018-02-19:
Please provide the ROM name and hash value:

**Copied and pasted from Issue template:**

> On Mac and Linux, use command line tools to get the hash values of your ROM (for example: `sha1sum`, `md5sum`). On Windows, you can use the [FCIV](https://support.microsoft.com/en-us/help/889768/how-to-compute-the-md5-or-sha-1-cryptographic-hash-values-for-a-file) utility.
> 
> If applicable, check against byuu's [SNES dump database](https://preservation.byuu.org/) and indicate if the bug occurs when using an original cartridge as well.

XOmniverse, 2018-02-19:
I've been playing a repro of the Final Fantasy VI - Ted Woosley Edition hack and noticed the same issue. I had chalked it up to the hack being weird. It occasionally happens whenever leaving a battle and is fixed by entering the menu/entering town/entering another battle/etc.

OP stated he's using the original cart so I doubt he can get an md5sum of the ROM.

saftle, 2018-02-19:
Ah you're right. I should have read the issue in its entirety.

masterninjah, 2018-02-22:
not sure if this helps but my version of the game is the one where you automatically pick up the cider after fighting a guy instead of having to pick it up off the table after the fight.  im guessing it would be revision 1 because of that.

frederic-mahe, 2018-02-22:
@masterninjah it seems that [byuu](https://preservation.byuu.org/games/Lj4b/3dDtNiEp) and [SNEScentral](http://snescentral.com/article.php?id=0043) only report one US version of the game (`SNS-F6-0`), no other revision.

masterninjah, 2018-02-22:
well my game is different than the one on youtube for that part and i have revision 1 of ff3 in my smokemonster 4.2 rom pack.  i couldnt even take the cider off the table, on the youtube video he takes it off and it disappears.  this shows what im talking about https://youtu.be/iGNskqYu2_0?t=10588.  i got a message saying i got the cider right after the fight and walking up to the table did nothing

frederic-mahe, 2018-02-22:
You are right, there is a Final Fantasy III (USA) (Rev 1) in SmokeMonster's pack `(SHA256:  10eccc5d2fab81346dd759f6be478dcb682eef981e8d3d662da176e1f9a996bc)`, but no Rev 0 `(SHA256: 0f51b4fca41b7fd509e4b8f9d543151f68efa5e97b08493e4b2a0c06f5d8d5e2)`.

It seems you have a cartridge that has not been referenced yet by SNEScentral :+1:. Could you open the cartridge and take a picture of the PCB? ([example](http://snescentral.com/pcb.php?id=0674&num=2&side=front)).

masterninjah, 2018-02-22:
lol cool, i just looked up a speedrun and they dont even run this version.  i could save about 3 seconds on their 5 hour run with this.

frederic-mahe, 2018-02-22:
thanks @masterninjah you have [this model](http://snescentral.com/pcb.php?id=0043&num=1&side=front). It is a Rev 0 as indicated on the largest chip (`SNS-F6-0`). That's a valid US cartridge.

SmokeMonsterPacks, 2018-02-25:
Please re-test this issue with the new official firmware v4.3 using an original cartridge. 

https://support.analogue.co/hc/en-us/articles/360000557452-Super-Nt-Firmware-Update-v4-3

masterninjah, 2018-02-27:
this issue seems to be fixed with the latest firmware.  im not 100% sure though because i think its random and doesnt happen every time.  i did the raft ride again without a problem but it was at a later point in the game without cutscenes where it happened so who knows.  this problem also happens after battles and i just did an hour and 15 minute level grind so i think its good.

ShinKaz, 2018-02-28:
@masterninjah just wanna chime in and say I had a similar bug yesterday with Marvelous (original sfc cart), looked extremely similar as you picture... could not reproduce it though. (was coming out of a tent like 15mins into a new game) it was with the jb fw though. (I did not give the newest official fw a shot yet)

SmokeMonsterPacks, 2018-03-04:
No follow up. Issue marked as resolved.

scotthammack, 2018-03-10:
I just encountered this on JB6.5 in the same area in Final Fantasy III (USA) (Rev 1), but before the Vargus fight. I just came out of a random battle on Mt. Kolts and the graphics were glitched. I went into the menu and back and they went back to normal.

neonstreetlight, 2018-03-16:
This graphical glitch had happened to me once when returning to the menu in Prince of Persia on Jailbreak 6.4. I haven’t seen it happen again, but it is extremely hard to replicate since it was so seemingly random.

SmokeMonsterPacks, 2018-03-27:
Closed until someone tests in JB v6.6. Please comment if it needs to be re-opened.

rumpleteazer, 2018-07-18:
Can confirm this visual glitch is still happening on  JB v6.6.

Had it once before boarding the raft from a save point and later after a battle. Both fixed by transition to another screen. I'm using an original cart.

buvk, 2019-07-19:
Just wanted to report that I just experience twice within about 30 minutes in FFVI Relocalization Project using JB v6.6.

pearljammzz, 2020-02-21:
I have been playing through FF3/6 on the latest firmware (4.9) and it still happens. Only happened to me on Mt Zozo in World of Ruin.

I am also seeing a lot of weird graphical glitches that I assume are connected. In certain instances characters will walk through/over the text box and this is most easily seen in the fight with Chadarnook in Owzer's basement. When Chadarnook is about to change forms you can see the new sprite form 'load' over the text box. Like I said, a very similar thing has happened in multiple cutscenes through the game but this one is easy to reproduce since there is a save right before. I can provide a save file if needed.

I am playing on a SD2SNES which is also on the latest firmware. Tested ROM on an original SNES and it doesn't have the issue that is present on the SuperNT.

bootsector, 2020-06-01:
This is what fixed Final Fantasy III on MiSTer SNES core:

https://github.com/MiSTer-devel/SNES_MiSTer/commit/7559d08dd58c6ad028eb8931753e3cdee50c6ea4

This basically fixes SDRAM's refresh.

Maybe the above could serve as an insight for a potential fix for the Super NT.

redacted173, 2020-07-31:
Not sure how useful this is, but I tried to hit the bug with a SD2SNES.  The problem in the intro I encountered is right after the fortress transitions to the scene with the overlook.  A DMA from $E26EC0 (ROM) to $0 (VRAM) of size $2000 during the screen transition with $420B write at $C0275B ends early.  The DMA ends before writing VRAM $006F (byte address) causing a large set of tile data to be from the previous scene.  It's interesting that the previous DMA to VRAM was probably from a save state being restored which used DMA channel 0 and was last reading address $F2FFFF.  This looks very similar to the extra address on the bus for the problematic DMA.

Another weird thing is the $420B write had a value of $9 on the bus, but in the code it looks like it should have been $1.  There were some weird transitions on the data bus from $01->$09->$01 on the LDA #$01 operand read.  But this may not matter.

Unfortunately, I couldn't make a short test that hits this which doesn't require custom software and SD2SNES modifications.  My cart extender is also of questionable quality, but I'm able to hit the same bug without it in.

Bad Tiles:
[image]

SNT Trace of DMA ending early:
[image]

SNT Trace of DMA running correctly:
[image]

1CHIP Trace of DMA running correctly:
[image]

redacted173, 2020-08-01:
The attached zip file contains a bps (floating) patch against ff3 US 1.1:
CRC-32: C0FA0464
SHA-256: 10ECCC5D2FAB81346DD759F6BE478DCB682EEF981E8D3D662DA176E1F9A996BC

It modifies ff3 (ff6) so it loops on the intro scene with the problem DMA.  It looks for the missing DMA and will hang using an infinite loop if the problem is detected by checking VRAM for the old (stale) data.  An 8b LFSR (seeded with the same value on reset) is used to produce a delay in frames prior to resuming execution for each loop iteration.  The LFSR is stored in the first byte of SaveRAM.  The loop works by reloading a full (minus audio) save state stored in the expanded ROM.

It's hit or miss as to whether it will actually have a problem but what I find speeds it up:
(1) Let the ROM run and wait 1 minute for the scene to cycle 5-6 times.  If this doesn't hit the problem go to (2).
(2) Try a game reset and go back to (1).  If this doesn't work then reset back to the SD2SNES menu by holding reset for a few seconds and go back to (1).

I can typically hit the problem in a few minutes by doing this.  It's possible, but rare for it to hit the problem by just letting it loop in (1) for a long time.

All options in the SD2SNES menu are disabled (in-game hooks, region patch, brightness patch, etc).  Not sure if this matters.

The trigger I normally use is continuous on ROM read from address $E26F2E, data value $54 which is the last successful read in the failed DMA.

[ff3us11_snt_issue27.zip](https://github.com/SmokeMonsterPacks/Super-NT-Jailbreak/files/5011149/ff3us11_snt_issue27.zip)

mattysaurus, 2020-12-04:
Just ran into this bug using the latest non-JB firmware and a Super Everdrive X5. Rom is No-Intro patched with latest version of Ted Woolsey Uncensored Edition.

re4mat, 2021-02-24:
Tested on JB7.0 using 1.1 ROM patched with redacted173's patch above. Did not hit the bug. (Have not tried with the vanilla ROM, though.)

closed 2021-02-24 by SmokeMonsterPacks