SmokeMonsterPacks/Super-NT-Jailbreak #28: Glitches: Ken Griffey Jr. Presents Major League Baseball (USA) (Rev 1)

barrusa, 2018-02-19:
## Expected behavior and actual behavior
Player names do not appear next to jersey numbers when they should otherwise be present. Reproduces on the team edit screen, lineup, and during gameplay in the corner when a player is up to bat.

## Visual examples
NOTE: Images were captured in `higan v106` and are meant to help illustrate where the issue exists when played on the `Super Nt`.

[image]
[image]
[image]

## Steps to reproduce the problem
1. Launch `Ken Griffey Jr. Presents Major League Baseball (USA) (Rev 1).sfc` game from SD card using jailbreak firmware.
1. Follow in-game menus to get to a game. Any mode will work, including the single game exhibition mode.
1. Observe the team lineups before the game begins, and the absence of player names in the corner during the batting phase of the game.

## ROM name and hash value
```
Ken Griffey Jr. Presents Major League Baseball (USA) (Rev 1).sfc
SHA256 checksum: F9F7A2DE8CBAFD6F9042841DFC42529F8245D75F436BED1C23F9BA1663182E61
```

(Matches [value](https://preservation.byuu.org/games/Lj4b/xH2utn9w) on byuu's [SNES dump database](https://preservation.byuu.org).)

## Firmware version and SD card model

- Jailbreak firmware v6.4
- SanDisk Ultra 32 GB SD card

SmokeMonsterPacks, 2018-03-04:
No follow up. Issue marked as resolved.

closed 2018-03-04 by SmokeMonsterPacks

scotthammack, 2018-03-04:
I just tested this on JB6.5 and am seeing player names, same as the higan screenshots. I never saw what it looked like broken but it appears to be fixed.

SmokeMonsterPacks, 2018-03-04:
Thanks, Scott!

barrusa, 2018-03-13:
Sorry for the delay -- I also tested this using the same ROM described in my original report, and my original cartridge of the game on the `JB6.5` firmware and as far as I can tell, the issue is resolved. Thanks!

SmokeMonsterPacks, 2018-03-13:
Thanks for the follow-up barrusa :D