SmokeMonsterPacks/Super-NT-Jailbreak #29: Samurai Spirits JP

Videodr0me, 2018-02-20:
Sumarai Spirits JP crashes after initial logo - sometimes cherry blossom animation shows for a fraction of a second before crash.

frederic-mahe, 2018-02-20:
@Videodr0me could you please indicate a hash value for the ROM? or are you using a cartridge?

Videodr0me, 2018-02-20:
SHA256: 208d65dbae14cc8a6d02df00528cb3d6cc67a3b84538a2141cdb949160239ecc
same file works on higan

SmokeMonsterPacks, 2018-02-25:
Could someone please re-test this issue with the new official firmware v4.3 using an original cartridge?

https://support.analogue.co/hc/en-us/articles/360000557452-Super-Nt-Firmware-Update-v4-3

SmokeMonsterPacks, 2018-03-04:
No follow up. Issue marked as resolved.

closed 2018-03-04 by SmokeMonsterPacks

Videodr0me, 2018-03-05:
Works on JB6.5

SmokeMonsterPacks, 2018-03-05:
Thanks for the follow-up, Videodrome