SmokeMonsterPacks/Super-NT-Jailbreak #31: Crash: Samurai Shodown (USA) (1.0)

macmenot, 2018-02-20:
## Expected behaviour

Game launches and plays

## Actual behaviour

Game glitches and then hangs just after loading and displaying initial title screen

## Steps to reproduce the problem

Launch game from SD card

## ROM name and hash value

[Samurai Shodown (USA) (1.0)](https://preservation.byuu.org/games/Lj4b/MDraApBe)

5db804171fca42486485ed85e4afe45b29e6d01304bdf75d520bfc42429739e3	Samurai Shodown (USA).sfc

## firmware version and SD card model

[snt_firmware_verJB6.4.bin](https://github.com/SmokeMonsterPacks/Super-NT-Jailbreak/releases/download/v6.4/snt_firmware_verJB6.4.bin)
[Samsung Memory 64 GB EVO Plus MicroSDXC](https://www.amazon.co.uk/gp/product/B00WIMC5IS)

frederic-mahe, 2018-02-21:
Thanks @macmenot This is possibly linked to the HDMA random crashes (see issue #33).

SmokeMonsterPacks, 2018-02-25:
Could someone please re-test this issue with the new official firmware v4.3 using an original cartridge?

https://support.analogue.co/hc/en-us/articles/360000557452-Super-Nt-Firmware-Update-v4-3

RevQuixo, 2018-02-25:
With the SD2SNES the game seems to work now under 4.3  It would previously crash at the intro.

SmokeMonsterPacks, 2018-03-04:
No follow up. Issue marked as resolved.

closed 2018-03-04 by SmokeMonsterPacks