SmokeMonsterPacks/Super-NT-Jailbreak #32: Cartridge crash on Link to the Past (USA)

siskavard, 2018-02-20:
I know most of the more important issues here are rom related, but I got a crash while playing my Link to the Past USA cartridge using the JB firmware.

It happened after completing the first major dungeon - Eastern Palace. Got the speedy boots from Sahasrala, went south to Lake Hylia and got my fortune read by the fortune reader guy. Upon exiting the fortune teller's the screen just stayed black indefinitely. I have not been able to reproduce it.

frederic-mahe, 2018-02-21:
Thanks @siskavard 

That could be related to the HDMA random crashes (see issue #33). That HDMA problem should be solved with the next official firmware release. Hopefully, the fix will be ported to the jailbreak firmware soon after.

SmokeMonsterPacks, 2018-02-25:
Please re-test this issue with the new official firmware v4.3 using your original cartridge. 

https://support.analogue.co/hc/en-us/articles/360000557452-Super-Nt-Firmware-Update-v4-3

scotthammack, 2018-02-25:
Assuming this is the same issue as the crash when going in and out of Link's house at the start of the game: I and some other people on the atariage forum have been trying to reproduce that on 4.3 and have not been able to make it happen yet. Hard to tell for sure since it's intermittent, but so far it seems like the fix was successful.

SmokeMonsterPacks, 2018-03-04:
No follow up. Issue marked as resolved.

closed 2018-03-04 by SmokeMonsterPacks