SmokeMonsterPacks/Super-NT-Jailbreak #33: Crashes Update

SmokeMonsterPacks, 2018-02-21:
Kev posted the following note, referring to his forthcoming update to the official firmware that addresses random crashes:

> I think many of those weird random crashes are a result of the "fix' I did for mecarobot golf that is on the 4.1 firmware.  I tinkered with the HDMA's and unfortunately it's possible for them to crash the system if things line up just right.  ninja warrior just happened to cause it to happen nearly continuously, hence the game crashing if you pause or just let it run a little while.   I did fix this bug (and a bunch of others) for the next release but I wanted to fix a few other things and add some new stuff before the next release.

http://atariage.com/forums/topic/242970-fpga-based-videogame-system/page-328#entry3967907

romevi, 2018-02-21:
Exciting stuff!

SmokeMonsterPacks, 2018-02-25:
Firmware update released: https://support.analogue.co/hc/en-us/articles/360000557452-Super-Nt-Firmware-Update-v4-3

closed 2018-02-25 by SmokeMonsterPacks