SmokeMonsterPacks/Super-NT-Jailbreak #36: Feature Request: Ability to Swap Functionality of Power & Reset Button on Console

cacophony1, 2018-02-24:
## Request

I'd like to have a toggle in the menu that swaps the functionality of the physical power and reset buttons on the console. So if enabled, pressing power will cause a reset and pressing reset will turn the console on/off.

## Reason

These two switches are the only movable parts on the console, and movable parts tend to wear out and fail over time. Nobody needs the physical reset switch, so if in the future the power switch fails it would be nice to be able to use the reset button instead.

masterninjah, 2018-02-25:
its a mechanical switch, they're usually rated for at least 10 million clicks.  good luck wearing that out.

cacophony1, 2018-02-25:
I've had a mechanical switch fail after around 1000 uses so problems do occur sometimes.

frederic-mahe, 2018-02-25:
@cacophony1 you are thinking long-term and I appreciate that. However, wouldn't it be better to replace the failing mechanical switch? I assume Analogue design's uses standard parts.

cacophony1, 2018-02-25:
@frederic-mahe well the switches are physically mounted onto the board so I assume that fix would require some desoldering/soldering skill (plus sourcing the correct part, which might be difficult): https://pbs.twimg.com/media/DWBchQoXcAMoIrH.jpg:large

SmokeMonsterPacks, 2018-02-25:
This is a nice suggestion but we're trying to focus the Issues tracker on bugs, so I have been closing feature requests to keep things focused.

closed 2018-02-25 by SmokeMonsterPacks

frederic-mahe, 2018-02-26:
@cacophony1 maybe Analogue can provide the specifications and references of the mechanical switches? You don't have much to loose in asking them.

cacophony1, 2018-02-26:
Good idea, it's worth a shot!