SmokeMonsterPacks/Super-NT-Jailbreak #38: Super Nt firmware update v4.3

frederic-mahe, 2018-02-25:
Analogue released a new [official firmware (v4.3)](https://support.analogue.co/hc/en-us/articles/360000557452-Super-Nt-Firmware-Update-v4-3).

romevi, 2018-02-25:
So this just flashes over the jailbreak, right? If I have the jailbreak installed, flashing the new official firmware won't cause any known issues, correct?

frederic-mahe, 2018-02-25:
yes @romevi you can flash any version of the official or jailbreak firmware at anytime. The Super Nt is designed to be "unbrickable". As long as you follow the flashing protocol, you should not have any problem.

joneslee85, 2018-03-02:
@frederic-mahe are we expecting a jailbreak version on top of v4.3?

frederic-mahe, 2018-03-04:
There is a new official firmware in town! (see issue #46)

closed 2018-03-04 by frederic-mahe