SmokeMonsterPacks/Super-NT-Jailbreak #40: Incorrect Width & Height for 1080p

Lemon-King, 2018-02-25:
## Expected behavior and actual behavior
I have an HDTV with overscan disabled, aka fill/full mode.
At 1080p resolution, height set to 1080 should fill the screen when the TV is set without overscan. However this isn't the case and height at 1080 will fill to where the edges where overscan ends.
This is weird and not the expected behavior I was expecting at 1080p resolution.

## Steps to reproduce the problem
Set your HDTV to full mode (not 16:9), and increase height.

## firmware version and SD card model
- stock firmware v4.3

cacophony1, 2018-02-25:
You're assuming that the game is using the full vertical resolution, which it might not be. Try the 240p test suite.

SegaSnatcher, 2018-02-26:
You have to set it to 1149 height if you want it to fill the screen vertically without overscan on a 1080p display.

Lemon-King, 2018-02-26:
That's still incorrect. A 1080p screen should fill at 1080 height, not 1149 which would be closer to 1200. A 16:10 height.

SegaSnatcher, 2018-02-26:
Thats because these old video games took overscan into consideration.  Its not a bug or a flaw with the Super NT, just how the games were programmed.   If you turn on overscan on your HDTV you'll see the image will fill the screen at 1080p height because of overscan compensation.  That is exactly what you would have experienced on a CRT since it stretches the 240p image.

Lemon-King, 2018-02-26:
Base settings for the Super NT 1080p 4.5x does not fill with overscan. It fills the screen to the borders where overscan would begin on an HDTV.

SegaSnatcher, 2018-02-26:
Why do you keep assuming its supposed to fill the screen at 4.5x?   Like Cacophony said, don't assume the games are always using the full vertical resolution.   What you are seeing is perfectly normal.  Just set your vertical height to +1149 and be done with it.

SegaSnatcher, 2018-02-26:
Framemeister works the same way, so again this is not a bug.

closed 2018-02-26 by SmokeMonsterPacks