SmokeMonsterPacks/Super-NT-Jailbreak #43: Boot to the last ran rom and favorites [Enhencement][Feature request]

ShinKaz, 2018-02-26:
Expected behavior
- add a "Title -> Last ran rom" and a "Last rom direct" option under "Startup Options" menu

- add a favorite option by pressing a button (start for example?) on any given rom in the SD file browser

then is would be displayed under Turrican 2 or Settings with the "star" unicode/utf-8 character (★) in front of it.
(press a key like start to remove it from the favorite list)

could look like : 

"Run cartridge
Super Turrican Director's Cut
Suuper Turrican 2
Settings

Favorites
----------
★My game1
★My game2
etc"

cacophony1, 2018-02-28:
Great idea! 

I would suggest just doing this in the top menu:

Run cartridge
Browse SD card
Browse Favorites    <------ new entry here
Super Turrican Director's Cut
Suuper Turrican 2
Settings

Browse Favorites would take you to a screen similar to the Browse SD card, but only containing the favorited roms. I think trying adding those roms to the root screen wouldn't be a as nice of a user experience for various reasons:

1. It adds too much clutter to the root screen
2. If the list gets long and the user has scrolled down, it won't necessarily be intuitive where the main root menu options are.
3. The font size of the root menu is much larger and not nearly as accommodating for long titles
4. Having the highlight jump over a "Favorites" header in a list as you navigate down is inconsistent with the rest of the UI

ShinKaz, 2018-02-28:
Makes sense, I do mostly agree, but the point was to be able to reach those favorites just like turrican1-2, without going through another menu. (about clutter, favorites could be limited to 5-10 or something)

@SmokeMonsterPacks > I feel the title was cleared with boot TO, "boot previous rom" kinda sounds like and option that could be under Run cartridge...which could be interesting too, but I'm really more interesting in directly booting to the last ran rom, when powering on... Sorry to if you consider this nitpicking, I'll shut up now. lol

B9KnM8UqQ2W2Vwk7uvZ7, 2018-02-28:
I'm wondering if it would be an idea to expand on this Favourites idea so that there's more than one customisable list? I'm thinking something along the lines of what you get on SD2SNES. Ignore the titles that I've used as you'd be able to create your own, but something like this.

 >PLAYLISTS
   >NOW PLAYING
   >SNES DRUNK PLAYLISTS
      >>BEST SNES FIGHTING GAMES
      >>BEST SNES MULTIPLAYER GAMES
      >>BEST SNES SHMUPS
      >>ENGLISH FRIENDLY SUPER FAMICOM GAMES
   >RETRO RGB PLAYLISTS
      >>BEST LOOKING SNES GAMES
      >>ESSENTIAL PLATFORM GAMES
   >FAVOURITE PLATFORM GAMES
   >FAVOURITE RPGS

This is something that I'm working on currently but at the moment I can only do it by having multiple copies of games when a game shows up more than once. So one copy of SMW would appear under AMERICA > S for me but a second copy will also have to be saved in the FAVOURITE PLATFORM GAMES folder and a third copy in the BEST LOOKING SNES GAMES folder.

If we were able to create playlists, or whatever they'd be called, we would only have a single copy of each game and it would tag them to appear in whichever folder we put them in.

I don't even know if this is possible but I thought it was worth mentioning here.

cacophony1, 2018-02-28:
@ShinKaz Maybe there could be an option for "Boot to favorites screen" as well? That way favorites could be separate from the root screen but users would have a way to select select them after boot without having to go down a level?

cacophony1, 2018-02-28:
@B9KnM8UqQ2W2Vwk7uvZ7 That sounds more like something the user would prefer to curate themselves via the folder structure they come up with in the regular roms directory. SNES roms are so small relative to the size of an SD card that saving space isn't of much benefit. I think the bigger problem is having most of these types of categorizations happen dynamically probably isn't technically feasible because on **every** boot the hash value of **every** favorited rom would have to be investigated to determine what folder it would need to appear in. That sounds like something a PC can handle but not SNES hardware.

But you maybe it would be possible to add dynamic folders like the following to the Favorites screen:
Last played
Most frequently played games <--- this could take to user to another screen that lists the top 20 played games, ordered by number of times launched

Those two items can be cheaply computed/stored each time a rom is launched.

cacophony1, 2018-03-01:
@ShinKaz Yeah, I'm guessing that a single save would work for all copies of the same rom but that's certainly a good question to bring up. I think it just matches the save file to rom based on the name so as long as the name is the same for all copies everything should be good.

ShinKaz, 2018-03-01:
yeah, that would make sense :)

honkphluxx, 2018-03-01:
The main issue I see with the current menu is the prominent position of turrican 1+2. In the end, those are also just roms. although stored on the super nt. After spotting the first game titles in the main menu, it is counter intuitive that the majority of games can be found behind a "Browse ..." item.

For me, it would already be an improvement, if the menu starts inside the sdcard folder.

cacophony1, 2018-03-01:
@honkphluxx Maybe another setting to "remove roms from main screen" that would hide the Turrican 1+2 entries?

B9KnM8UqQ2W2Vwk7uvZ7, 2018-03-01:
@ShinKaz 
That's my main concern with having multiple game saves.
I haven't set up my folder system on it yet with multiple copies of the same game to see how well that would work but it's also probably worth waiting until we're past the current saving issues in the jailbreak firmware.

If multiple copies of the same game in different folders do work with a single save without any issues then I don't really think there's any advantage in doing what I suggested, the ROM size is just so small in relation to the capacity of SD cards that's not really going to be an issue for people. if it was, you can buy a 32GB for almost nothing and even if this eventually supported all the Nt mini cores it's still not going to be an issue for people.

As long as the saves work properly then it probably wouldn't even be worth the effort to do what I mentioned. 

@honkphluxx @cacophony1 
Maybe I shouldn't ask but is the Director's Cut available outside of the Super Nt now? I'm just thinking that if it is people could hide the two games on the main menu and then put them wherever they want on their SD card. Obviously you can do that with the second game but I wouldn't really want to lose access to the Director's Cut because I do enjoy the game.

SmokeMonsterPacks, 2018-03-04:
Closing feature requests that weren't addressed.

closed 2018-03-04 by SmokeMonsterPacks