SmokeMonsterPacks/Super-NT-Jailbreak #45: 1:1 setting should simulate square pixels, not an aspect ratio

jaysylvester, 2018-03-04:
## Expected behavior and actual behavior

The term "1:1" used by the retro community commonly refers to pixel ratio (square pixels). Seeing this label, the user working with the community definition assumes square pixel simulation by utilizing the same multiplier in both directions (240x256 -> 1200x1280, so 5x in both directions).

However, the Super Nt's 1:1 setting represents an aspect ratio and makes height and width the same (1200x1200 display pixels), which doesn't simulate square pixels.

The expected behavior is that using the 1:1 setting would use the same multiplier for horizontal as used in the vertical height setting.

There was also some discussion about changing the label on this setting to something like "Simulate Square Pixels" since every other ratio in the UI (4:3, 16:9) refers to aspect ratio, making 1:1 a confusing label to those unfamiliar with its usage in the retro community.

## Steps to reproduce the problem

Enable 1:1 in the menu with any vertical multiplier. Width will be set to the same display pixel value as vertical rather than an equal multiplier.

## firmware version and SD card model

Affects all versions.

jaysylvester, 2018-03-04:
"Aspect ratio" anywhere else in the video chain refers to display aspect ratio, not pixel. I know the retro community calls it that when referring to pixels, but any typical user would find that description confusing.

cacophony1, 2018-03-04:
I don't care what it's called, but practically speaking there are some people who want square pixels, but I don't think anybody is interested in a square display ratio.

jaysylvester, 2018-03-04:
The main reason I'm suggesting a label change is because the label is for the "easy" standard menu, which is what less technical users will work with. The enhanced settings make it possible to set a custom 1024x960 or 1280x1200 size if that's what someone wants, and I'd assume advanced users have already figured that out if it's important to them.

It should be evidence enough that the term "1:1" is confusing since kevtris himself took it to mean an aspect ratio and not pixel ratio, and he's a pretty smart dude.

cacophony1, 2018-03-04:
Makes sense. I like the idea of calling it "square pixels" to avoid confusion.

frederic-mahe, 2018-03-05:
That's a request for the official firmware. This is an interesting discussion but I am going to close this issue as it does not concern a jailbreak feature in particular.

closed 2018-03-05 by frederic-mahe

jaysylvester, 2018-03-07:
We've been told to report issues with the official firmware here as well. Is there somewhere else we should be suggesting improvements other than Discord/AtariAge?

jaysylvester, 2018-03-08:
No problem, just wondering. Thanks for answering 👍