SmokeMonsterPacks/Super-NT-Jailbreak #47: Super NT jailbreak firmware update v6.5

frederic-mahe, 2018-03-04:
A new jailbreak firmware is available! Many thanks to those who reported bugs with the previous firmware. 

Changelog:
  - adds a new "Save?" dialog box,
  - fixes DMA-related bugs, which should solve most random crashes,
  - fixes both SRAM bugs (firmware 6.4 only saved once and had a nasty
    corruption bug). Please test it again so we can be sure the save
    functionality works as expected. Any save games made on the
    jailbreak firmware 6.4 should be considered as potentially
    corrupted and must be discarded,
  - fixes glitch with some games disabling the "return to menu"
    shortcut

B9KnM8UqQ2W2Vwk7uvZ7, 2018-03-04:
Fantastic, I've not tried v4.4 yet but can I assume that this includes all the improvements from the "official" version?

frederic-mahe, 2018-03-05:
@B9KnM8UqQ2W2Vwk7uvZ7 added to the changelog, as suggested.

SmokeMonsterPacks, 2018-03-11:
Should we keep news posts limited to a week to tidy up?

frederic-mahe, 2018-03-11:
@SmokeMonsterPacks ok!

closed 2018-03-11 by frederic-mahe