SmokeMonsterPacks/Super-NT-Jailbreak #48: "Save RAM" confirmation dialog stops appearing if you leave the SD card browser

scotthammack, 2018-03-04:
## Expected behavior and actual behavior
Expected behavior: User is presented with the confirmation dialog any time they bring up the menu.
Actual behavior: The confirmation dialog only appears if the menu is opened while in the SD card browser submenu. If the user has backed out of that menu into the main system settings, the confirmation dialog will not appear and it becomes impossible to write the save to the SD card.

## Steps to reproduce the problem

- Boot Super Nt.
- Browse SD card and run any ROM.
- Use the menu hotkey to bring up the menu.
- Use the cancel button to back out through the menus until you're out of the SD card browser and back in the main Super Nt menus.
- Close the menu.
- Open the menu again.

You will not be presented with the confirmation dialog to write the SRAM to the SD card. The only way to get it back is to go back into the SD card browser and start another ROM (but by doing so, you lose your opportunity to save on the one you had loaded initially).

## ROM name and hash value
Any ROM.

## firmware version
- Jailbreak firmware v6.5

iancourt, 2018-03-04:
Confirmed.

-- 
idc

cacophony1, 2018-03-05:
Maybe an easy fix would be to have the root SD browse screen prevent using the BACK/CANCEL key to return to the main menu?

Llorean, 2018-03-05:
This is the same issue as Issue #12

scotthammack, 2018-03-05:
Also of note: the workaround doesn't allow you to get to the Tools menu, so if you want to toggle codes during a game, you have to sacrifice the ability to save.

cacophony1, 2018-03-05:
@Llorean I wouldn't call this the same issue #6. It has some overlap and it may be caused by the same thing, but the too user facing bugs here are quite different (not being able to save vs not being able to return to game)

Llorean, 2018-03-05:
They really are both the same issue. The steps to reproduce are the same. Literally this bug here is "when issue #6 happens, the new save prompt doesn't display anymore either."

Llorean, 2018-03-05:
Even in the previous firmware version, #6 would prevent saving.

cacophony1, 2018-03-05:
Just because two bugs have the same repro steps doesn't they should be treated as the same bug. When you call something the same bug you usually dupe it, which I don't think makes sense in this case.

You're speculating they have the same underlying cause, and that one is fixed there other one will work with no additional work.  This may be true. Or it may not be. Unless you're the developer you can't possible know for certain.

cacophony1, 2018-03-05:
@Llorean No it isn't. The title doesn't mention it. The original report doesn't mention it. None of your comments mention it. The only mention is in the single quote about a potential workaround.

@frederic-mahe I think all of this potential dupe discussion should be removed from this bug as it adds needless clutter

Llorean, 2018-03-05:
If a series of steps causes multiple problems vs. one problem vs one problem, it's still a single bug with multiple vs. single symptoms. If anything more info on the save failing should be moved there.

cacophony1, 2018-03-05:
Well we'll have to disagree then. I think that adding it to other bug will only make it more confusing to the person actually fixing the bug(s) and will just increase the likelihood of something being missed.

frederic-mahe, 2018-03-05:
@cacophony1 @Llorean you've both presented valid arguments. Each issue was reporting a different problem initially. Now we know more, it turns out this root of the problem may be the same in both cases, and a fix may solve bug issues simultaneously.

I'll maintain the `duplicate` tag (it should be `near duplicate` in reality) and I'll reference the issue 48 in the flow of issue 6 to reflect that. And I will not erase your comments (again, both of you raised interesting points).

worleydl, 2018-03-05:
Not sure if it's worth raising a separate issue but I experienced this behavior loading Secret of Mana. Down + Select quit working unless I pressed B in game, at this time Select/Start were not read in game.  I feel like the system menu is maintaining hooks to the input after being dismissed but that's just an uneducated guess.

worleydl, 2018-03-05:
To add, sometimes the inputs while I was in the menu were forwarded to the game. Seems like there's a state sync issue.

neonstreetlight, 2018-03-07:
Experience this same issue. Once back in the game, it actually loads the menu by pressing non-hotkeys. This would be less of an issue if there were an option to automatically save the SRAM (similar to the Analogue Nt Mini). Not being able to trigger the save SRAM prompt causes a total loss of any saves that occurred after the last SRAM save.

cacophony1, 2018-03-07:
@neonstreetlight Had the auto save been in place it would likely have not saved as well, it's just you wouldn't have known about the issue until you came back later to resume your progress. The confirm dialog was added specifically as a way to confirm that the save is going to take place so that bugs like this one could more easily and quickly be identified.

SmokeMonsterPacks, 2018-03-14:
I've added this to "Known Issues" in the [README](https://github.com/SmokeMonsterPacks/Super-NT-Jailbreak/blob/master/README.md).

SmokeMonsterPacks, 2018-03-27:
This should be resolved in the new JB v6.5 firmware.

closed 2018-03-27 by SmokeMonsterPacks