SmokeMonsterPacks/Super-NT-Jailbreak #49: Various BS Zelda Hacks Glitch/Freeze on start

cacophony1, 2018-03-05:
Various BS Zelda Hacks don't work via SD card

## Expected behavior and actual behavior
Using my Super Everdrive V2 with my Super Nt and latest JB6.5 I'm able to register a character and play the following 3 roms fine:
- BS Zelda Fourth Quest Map 2 Eng (Hack) T+Eng Final Roto, Con, DreamerN, DSerkol
- BS Zelda Link Version Map 1 (Hack) All Optional Patches
- BS Zelda Third Quest (Hack) red hearts
(all three of these roms are in the Smokemoster pack in the /3 BS-X Satellaview/BS Zelda directory)

None of these three roms work on the Super Nt if I load them directly via Browse SD Card. In all three cases on the character "SELECT" screen I'm presented with three already filled out options, all with the name "QQQQQQQQQ", with 6 hearts and and "30" (whatever that means). And when you select one of those characters you get taken to a glitchy screen (see screenshot below) as music plays in the background. At this point none of the buttons do anything and the screen remains static.

[image]

frederic-mahe, 2018-03-05:
Thanks for that detailed bug report @cacophony1

leodsgaming, 2018-03-06:
Tested the bug, it shows exactly as described, same charaters on screen and all. You can still exit to the Firmware menu with down+select.

SmokeMonsterPacks, 2018-03-27:
Closed until someone tests in JB v6.6. Please comment if it needs to be re-opened.

closed 2018-03-27 by SmokeMonsterPacks