SmokeMonsterPacks/Super-NT-Jailbreak #50: Improvement: True 60.0988 Hz HDMI Output

Videodr0me, 2018-03-05:
Super NT video is not "totally accurate". Actual behaviour: 

1) In zero-delay mode the super-nt is slowed down to 60hz instead of 60.0988hz. This makes all games run to slow.

2) In fully-buffered mode the super-nt runs at correct speed, but frames are dropped leading to jerky scrolling.

3) In unbuffered mode the super-nt runs at correct speed, but exhibits tearing artifacts.

While the current behaviour is a good compromise to achieve compatibility with all HDMI TV/Monitors, a large number of HDMI TVs and Monitors are able to display slightly increased scan rates. 

Proposal:
an additional HDMI mode close(r) to the 60.0988hz. It should not replace current modes but be an entirely additional optional mode for users with a compatible TV. Even if activated accidentally on not compatible TVs, it should not pose a problem as starting with Firmware 4.4 (JB 6.5) holding RESET while POWERING on reverts to a compatible HDMI signal. 

Even if this is not achievable by increasing the HDMI clock, it could be achievable by reducing H-Sync/H-Front-Porch/H-Back-Porch and/or V-Sync/V-Front-Porch/V-Back-Porch values. Maybe a buffer at SNES resolution of a small number of scan lines (depending on picture position) might be nessecary.

Affects all Firmwares upto and including 4.4 and Jailbreaks upto and including 6.5

ShinKaz, 2018-03-05:
regarding this issue, all my hopes a directed towards future solutions like the Analogue dongle+ a Variable FrameRate HDMI 2.1 conpatible OSSC. (or a future Analogue console that'll have VFR... FPGA NeoGeo maybe ? and the SNES core)

I hope there's an easier to achieve this though..

SmokeMonsterPacks, 2018-03-05:
Loads of users have requested this and Kev said it's not possible in some streams. It's an official fw request, and will either happen or not regardless of feature creeping.

closed 2018-03-05 by SmokeMonsterPacks

Videodr0me, 2018-03-05:
How is this feature request different from others? If "loads of users" have requested this, isn't it even more important for this to remain open as a reminder. Yes, this also pertains to the official firmware, but so do many of the bugs reported here. This is the only systematic tracker for issues and feature requests, if there is a place for it, isn't it here?

As for Kevtris saying it's not possible - that sounded to me more like "it's not simple, and most certainly not now", but its very hard to believe it is "impossible". I am not suggesting that this should have any priority and i still love the super nt, even if such a mode never materializes , but it's the one major thing that keeps it from really being "total accurate" and shouldn't be forgotten.

frederic-mahe, 2018-03-05:
@Videodr0me we are trying to focus on issues that concern the jailbreak firmware usage (navigating and playing games stored on an SD card, and saving to an SD card rather than on a cartridge). Reporting issues on other aspects of the Super Nt experience is accepted, but we will close these issues eventually. Now, **closed issues are not erased**. The developer of the jailbreak or the people at Analogue can use the tags to browse and filter them at anytime. Your ideas and suggestions won't be lost.

Videodr0me, 2018-03-06:
Ok, this is good enough for now. Is is ok to nag again in let's say two month?

frederic-mahe, 2018-03-06:
Feel free to make that issue re-surface periodically, but don't blame us if we close it again :-). In a couple months, firmware coders may have more time for new features.

Hovushka, 2019-01-11:
As far as my knowledge goes, original refresh rate is still unsupported...