SmokeMonsterPacks/Super-NT-Jailbreak #51: Issue: No Hot-Keys in Zootoo Mahjong (JP)

Videodr0me, 2018-03-05:
Zootoo Mahjong (JP) disables all hotkeys.
JB 6.5

leodsgaming, 2018-03-06:
Bug tested, Exactly as described. No hotkeys work.

leodsgaming, 2018-03-06:
On the game Spriggan Powered no hotkeys work on either JB or the official Firmware.

On the game Dynamic Stadium the Hotkeys work in certain moments in the game, and don't work in others.

Secret of mana is the most odd of them all:
JB but through SD2SNES you can summon the menu in all situations unless P2 opens his inventory. If he does, then P1 can't access the menus. But on the Jailbreak, you need to open your inventory or enter dialogue for the hotkeys to be usable.

Reinforcing: The behaviour is different when the game is side-loaded through the SD browser, or loaded through SD2SNES.

SmokeMonsterPacks, 2018-03-06:
Thanks, @leodsgaming. I've renamed the thread to reflect this as a more widespread issue. Please continue to list any games where this problem occurs.

Scythesurge, 2018-03-12:
I noted this issue on a couple other games, notably The Peace Keepers/Rushing Beat Shura. Hotkeys still worked while at the title screen & menus but once in game the hotkeys stopped working. Hitting the reset button to return to title allowed hotkeys to be used again at the title/menus but not in game. Another game I noted the issue in but could not reproduce was Pieces.

cacophony1, 2018-03-14:
Everyone having these issues should specify whether they occurred using an 8bitdo wireless controller, and if so, whether it also occurs with a wired controller.

There's currently a discussion on discord about NBA Jam not being responsive to keys at certain times and it became apparent that the only people having issues were using 8bitdo controllers. Kevtris suspects the issue might be with the 8bitdo firmware in this case.

scotthammack, 2018-03-14:
I did some testing on various games mentioned here on both the 8bitdo SN30 and an original SNES controller. Short version is the OEM controller worked in every case I tried it, but the 8bitdo acted the way people described in the bug reports.

Here are the full results: 
NBA Jam: hotkeys work on OEM, do not work on 8bitdo
The Peace Keepers: hotkeys work on OEM (both on the title screen and in-game). On 8bitdo, work only at certain times (on title screen, but not copyright screen, character select screen or in-game)
Spriggan Powered: hotkeys work on OEM, do not work on 8bitdo
Dynamic Stadium: hotkeys work on OEM at all times as far as I can tell. On 8bitdo, work only at certain times (after hitting a baseball they would work briefly during the scene of watching where the ball goes)
Secret of Mana: hotkeys work on OEM at all times as far as I can tell. On 8bitdo, works as leods described (only in menus or when there's a dialog box on the screen)
Zootto Mahjong!: hotkeys work on OEM, do not work on 8bitdo

All of these were tested on JB6.5, loading directly from the SD card (no SD2SNES or anything).

The 8bitdo controller is the SN30 that they were selling to go with the Super Nt (the one that looks less like a real SNES controller). Firmware on the receiver is whatever it shipped with; I haven't upgraded it.

frederic-mahe, 2018-03-14:
@scotthammack thanks for this thorough testing!

Videodr0me, 2018-03-14:
This is good news. I only tested with the 8bitdo wireless receiver. As this is the official controller analogue recommends, chances are high that 8bitdo will do a firmware update. Does anybody know if analogue is aware of the 8bitdo issue and in talks with 8bitdo to solve the issue?

Videodr0me, 2018-03-14:
I upgraded the 8bitdo retro receiver from 1.21 to 1.30 and the controller from 4.01 to 4.10. Hot-keys still do not work in Zootoo. So we have to wait for a newer firmware from 8bitdo!?

SmokeMonsterPacks, 2018-03-14:
Maybe we can now close this issue and just add a note in the README about the 8bitdo controllers?

cacophony1, 2018-03-14:
@SmokeMonsterPacks That makes sense. IIRC, kevtris mentioned has was going to bring this up with 8bitdo, so probably no reason to keep this open here.

closed 2018-03-14 by SmokeMonsterPacks

SmokeMonsterPacks, 2018-03-14:
I've added this to "Known Issues" in the [README](https://github.com/SmokeMonsterPacks/Super-NT-Jailbreak/blob/master/README.md).

t3rminus, 2018-03-29:
**Super Gameboy 2** - With the Super NT hotkey set to `start`+`select`, pressing the Super Gameboy 2 menu combo (`L`+`R`) also triggers the Super NT menu for some reason, only with the 8bitdo SN30/Retro Receiver combo.

This was originally noted in #95

ChocoChopin, 2022-07-09:
I'm definitely having this issue in both versions of Super Puyo Puyo Tsuu, using the 8bitdo Retro Receiver on latest 1.39 firmware. The problem persists whether I'm using the 8bitdo SN30 or the official Switch SNES controller. I'm struggling to wrap my brain around how this bug could even work--what should it matter whether the inputs are coming from a physical wired controller or the retro receiver? How can the Super NT "tell" such that it only bugs in this way with the retro receiver?