SmokeMonsterPacks/Super-NT-Jailbreak #52: wishlist: support for pseudo symbolic links

macmenot, 2018-03-05:
Rather than storing multiple copies of the same `.sfc` on the SD card, it would be useful if we could employ symlinks to allow multiple path entries to actually point at the same underlying rom file. 

Obviously FAT32 doesn't support real symlinks, but we could presumably add support for a plaintext file containing nothing but a a path to the real underlying `.sfc` on the same SD card. This would allow us to have the complete SMDB directory structure and entries on the SD card without needlessly duplicating the rom files multiple times.

The SMDB contains 9809 individual rom path entries, but only 5286 unique checksums :
```
 % cut -f1 < Super\ EverDrive\ \&\ SD2SNES\ SMDB.txt | wc -l
    9809
```
vs
```
cut -f1 < Super\ EverDrive\ \&\ SD2SNES\ SMDB.txt | sort -u | wc -l
    5286
```

SmokeMonsterPacks, 2018-03-05:
Same as #43

closed 2018-03-05 by SmokeMonsterPacks

macmenot, 2018-03-05:
Hmm, considering #43 was already closed as (essentially) wontfix I'm not really sure I'd consider this the same request. #43 described harder-to-implement support for playlists and editing of custom layouts and favouriting within the firmware and menus, and I assume was dismissed because of that. Symlink support would be a far simpler implementation, confined to the rom loading logic, but would allow a path redirect after reading a file once.

frederic-mahe, 2018-03-05:
@macmenot what you suggest was discussed briefly in #43. The consensus for now is that the benefit-cost ratio is not high enough to implement file deduplication. ROMs for 8-bit and 16-bit games are small and SD cards are getting cheaper and larger (roughly 10 times larger in 10 years).

File duplication is a real issue for collections of CD-ROM and DVD-ROM games, which could be solved by using filesystems supporting hard links (i.e. not FAT32). You've suggested an interesting workaround, but that's out of the scope of the jailbreak firmware for now.