SmokeMonsterPacks/Super-NT-Jailbreak #53: Kaite Tsukutte Asoberu Dezaemon (JP) fails with s-ram break down

Videodr0me, 2018-03-06:
After Kaite Tsukutte Asoberu Dezaemon (JP) displays the company logo it does a self check (graphic, map, data, music, s-ram). It fails that test and displays "s-ram break down" and just hangs.

Interestingly the fan translation  "Kaite Tsukutte Asoberu Dezaemon (J) T+Eng1.0 AGTP" immediatly goes into an infinite loop displaying "DEZAEMON SUPER FAMICON VERSION" over and over again.

All on Jailbreak 6.5

SmokeMonsterPacks, 2018-03-06:
Thanks. Please post hashes of the ROMs you're using.

frederic-mahe, 2018-03-06:
@Videodr0me can you please confirm you are using the ROMs below?
```
Kaite Tsukutte Asoberu Dezaemon (J) T+Eng1.0 AGTP.smc
    b2ded10e06c5dbe338be9e10d2dfb4d5ca8ada01157cbfc71bae20ae107ad48d
    d49cfc56819a0075a095e921ae4c5b781ff80c8e
    a914569f7a0ace21c37351b6e0e49f6c
    e3067b47
Kaite Tsukutte Asoberu Dezaemon (Japan).sfc
    3ddf81cee32dcf7c8df4367bdae5ea1b0af50b7baab54be1c01f6ae6c3e308a6
    4b5725aeb7f770feadda6cfec5e609ff57fce363
    e6f13860e50b921ec09a89fd5b585ac7
    52167cc
```

leodsgaming, 2018-03-06:
Sorry, I am really new to this. What is the crazy string of characters, how do I find that?

I tested this bug and it presented itself exactly as described in the report. ROM name Matches too.

frederic-mahe, 2018-03-06:
@leodsgaming On Mac and Linux, you can use command line tools to get the hash values of your ROM (for example: `sha1sum`, `md5sum`). On Windows, you can use the [FCIV](https://support.microsoft.com/en-us/help/889768/how-to-compute-the-md5-or-sha-1-cryptographic-hash-values-for-a-file) utility.

Videodr0me, 2018-03-07:
Kaite Tsukutte Asoberu Dezaemon (Japan).sfc
sha256: 3ddf81cee32dcf7c8df4367bdae5ea1b0af50b7baab54be1c01f6ae6c3e308a6

Kaite Tsukutte Asoberu Dezaemon (J) T+Eng1.0 AGTP.smc
sha256: b2ded10e06c5dbe338be9e10d2dfb4d5ca8ada01157cbfc71bae20ae107ad48d

Also tried them on Higan - and both show the exact same behavior as on the super-nt! Can somebody with an sd2snes and - even better - with an original cart try this title?

frederic-mahe, 2018-03-07:
byuu announced a [fix for Higan](https://twitter.com/byuu_san/status/971456653315932161), and communicated with kevtris. Let's hope the issue can be fixed upstream in the official Super Nt firmware.

SmokeMonsterPacks, 2018-03-27:
Great Hierophant confirmed that this is resolved in v6.6

closed 2018-03-27 by SmokeMonsterPacks