SmokeMonsterPacks/Super-NT-Jailbreak #54: Improvement: add HDMI CEC support

strabbit, 2018-03-06:
**REQUESTED FEATURE**
Add an option to the menu to enable HDMI CEC auto-switching.

**REASON**
So that upon powering on the Super NT the television automatically turns on (if it wasn't previously) and switches to the correct source.

CoolMatty, 2018-03-10:
Also, additionally, this would allow home theater receivers to recognize it and present it as a special input, drastically improving usability with such equipment. For example, the PS4 will appear as "Playstation 4" on the input switcher, instead of requiring you to go to the receiver then change receiver inputs.

SmokeMonsterPacks, 2018-03-19:
The CEC line in the SNT isn't connected, so this won't be possible.

closed 2018-03-19 by SmokeMonsterPacks