SmokeMonsterPacks/Super-NT-Jailbreak #55: Problem with Starwing (PAL, German) Cartridge.

helohe, 2018-03-06:
## Expected behavior and actual behavior

Whenever 3d graphics are shown a vertical line of artifacts runs down from top to bottom of the screen.
Titlescreen and menus look fine.

## Steps to reproduce the problem
Just play the game with the cartridge, the problem appears immediately.

## firmware version and SD card model

- Jailbreak firmware v6.5
- console set to PAL (tried NTSC also, no difference)

Altazimuth, 2018-03-07:
My UK PAL copy (SNSP-FO-UKV) runs fine when the console is set to PAL. Not sure what difference the two regions would have though (outside of just text).

SmokeMonsterPacks, 2018-03-07:
Report back after testing in 50hz mode please.

frederic-mahe, 2018-03-07:
So the bug is not confirmed yet. The German version is localized (see [SNSP-FO-NOE](https://preservation.byuu.org/games/piDW/ihRB3hnk)) There are two UK revisions of Starwing `SNSP-FO-UKV`: `SPAL-FO-0` and `SPAL-FO-1`, both are distinct from the German version (different ROMs). Maybe this is specific to the German version?

It could also be a dirty contact on the cartridge. @helohe could you please try with the ROM of Starwing (Germany) rather than with a cartridge?

leodsgaming, 2018-03-07:
A used reported on Discord, that setting the video to 50hz solves the problem. You need System set to PAL and video to 50hz for this game to work as intended apparently.

The ROM can't be played, it's a Super FX game. Not even SD2SNES can play that.

helohe, 2018-03-07:
50 HZ and PAL does indeed solve the problem for me. Best thanks.

Videodr0me, 2018-03-07:
Recommendation: 
1) Put HARDWARE SETTING (PAL/NTSC) and VIDEO MODE (50/60HZ) on the same setting page.
2) Whenever someone changes Setting to PAL or NTSC, preselect corresponging VIDEO MODE (50hz for PAL, 60hz for NTSC). User can then if he reallly really wants to play a PAL game in 60hz set VIDEO to 50hz again.

Otherwise and without an auto mode - many bugs will be reported such as this, which are in fact not bugs but arise because of the PAL/NTSC 50/60hz mismatch.

frederic-mahe, 2018-03-07:
@leodsgaming my bad, I forgot this is a special chip game.

B9KnM8UqQ2W2Vwk7uvZ7, 2018-03-07:
Glad it's been resolved.

There are some PAL games out there that were "optimised" for PAL systems prior to release so they usually don't play well running as NTSC software. When you combine that with a special chip it can really amplify the issues that you may have.

From reading the description, it is a recognised issue with this particular title when running it on NTSC machine.

ShinKaz, 2018-03-08:
yeah...don't link PAL to 50 and NTSC to 60hz though...

We PAL user have been aching for running PAL at 60Hz like on real switched PAL snes since the Super NT launch.
only 5% (give or take) where PAL50 optimised, we need an option for the rest.

SmokeMonsterPacks, 2018-03-08:
@ShinKaz Thanks. So the way it is now is ideal for you? If so, I will probably close this issue.

frederic-mahe, 2018-03-08:
I think @Videodr0me offered a balanced solution (good parameters are on the same page and pre-selected) that can avoid frustration to non-savvy users, and still alow more advanced users to do what they want. Therefore, I vote to maintain the issue opened @SmokeMonsterPacks

SmokeMonsterPacks, 2018-03-08:
That sounds like a good compromise, and easy to implement. So I agree this should stay open.

ShinKaz, 2018-03-08:
@SmokeMonsterPacks it's ideal for it to stay separated, same page would be good too.

but more importantly, I need PAL games to REALLY run at 60Hz when set to 720p60 or 1080p60, which they aren't for now. (it's just running at 50Hz with added frames and looks horrible, since launch and most ppl don't even notice, or care...I mean for me, this goes beyond the jailbreak fw, I want to play a game at 60Hz ? fine, I can run the NA or JP rom instead... but what about running genuine carts fur us in europe ? honestly, if I can get my snes to do this with a switch between ground and a pin, I'm pretty sure it's less than 5mins of work on the HDL source. yet, the super NT has been out a month and had WAY more complicated fixes involving timings and DMA. that really worries me, tbh.)

https://shmups.system11.org/viewtopic.php?f=6&t=61037&sid=9971373056c0298b2707d88e83c37d79&start=480

SmokeMonsterPacks, 2018-03-08:
@ShinKaz I suggest writing that up as an issue if you want it addressed. Minus the part about it being 5 minutes of coding.

Videodr0me, 2018-03-09:
@Shinkaz: If I understand your issue correctly you claim:
If Super NT is set to "Hardware -> PAL and Videomode->60hz (any resolution)" then the games do not behave like running on a PAL consoles modded for 60hz. If this is true (and i did not confirm yet, as i am mostly interested in JP games) this is a completely different issue and I agree with SmokeMonsterPacks that you should write this up as a seperate issue.

gnadenlos, 2018-03-19:
PAL60 (real 60fps/Hz and PAL region code) should be a possible choice. I don't really care how exactly it is implemented. The easiest way would be to offer PAL50, PAL60 and NTSC60 instead of PAL and NTSC.

tltx, 2018-03-20:
I'm not sure but this is how I think it works. Hardware -> PAL means running the console in 50 Hz internally and Hardware -> NTSC in 60 Hz internally just like a real SNES would change speed when you set PPU1 pin 24 and PPU2 pin 30 hi or low with a mod. This is an accurate simulation of the hardware and maybe you have to change the speed to have games recognize it as PAL or NTSC.  Videomode ->1080p60 for example is what you want the console to output via HDMI and not a part of the simulation but a setting of the built in video scaler. You want to match the internal speed with the HDMI output speed so the you don't have to introduce extra frames or drop frames. Most games was made to run in 60 Hz even a lot of PAL games is 60 Hz  games just slowed down to 50 Hz giving us in Europe a lesser experience (e.g F-Zero) so you typically want to run games in Hardware -> NTSC, Videomode -> 1080p60 if possible. Some games have a region lock that somehow checks if the hardware is running in PAL or NTSC mode but most of them only check this at boot up and can be fooled by starting with  hardware -> PAL and swapping to NTSC as soon as you see the title screen, you don't have to change the actual HDMI output speed as they are not related, doing that would only make the HDMI renegotiate mode and have the tv go to black for a second and show input info. Some games, like Yoshi's Island PAL is actually made especially for PAL and just runs bad in NTSC so you want to run them in PAL and to have a matching HDMI output that does not add extra frames, so set the videomode to 1080p50. If possible it would be great to have a feature that fools games that they are running on PAL hardware even thought they are actually running in 60Hz internally (PAL60?).

SmokeMonsterPacks, 2018-03-28:
Kev wants to smooth out region issues in a future official fw, so I'm going to close these to get the focus back on bugs.

closed 2018-03-28 by SmokeMonsterPacks