SmokeMonsterPacks/Super-NT-Jailbreak #56: Improvement: add “Restore AC Power Loss” option

Locutus73, 2018-03-07:
**REQUESTED FEATURES**

An option to set the console behavior after power outage. Similar to the option found on PC BIOSes. The values should be:
- Power Off (default value, current behavior)
- Power On (the Super NT should automatically power on as soon as it gets power from the transformer)
- Last State (nonessential option, the Super NT should return to the last power state)

**REASON**
Using “Power On” value the Super NT could be powered on and off remotely using a domotic power socket. I.e. I use all my gaming gear through wireless controllers and HDBaseT extenders in different rooms and I turn them on and off through the controllers or domotic sockets.

TL;DR: Hey Siri, turn on the Super NT!

SmokeMonsterPacks, 2018-03-07:
Interesting idea. You might want to message Kev on discord to find out if it's technically possible, since it's definitely an official fw concept.

SmokeMonsterPacks, 2018-03-07:
Super NT has a physical power switch, so this is impossible.

closed 2018-03-07 by SmokeMonsterPacks