SmokeMonsterPacks/Super-NT-Jailbreak #57: Audio bug in Arabian Nights - Sabaku no Serei Ou (Japan)

cacophony1, 2018-03-08:
I'm posting this issue on behalf of a couple people on AtariAge.

Melter said:
"I think I found a strange audio bug in Arabian Nights - Sabaku no Serei Ou (Japan). During the intro cutscene, a few mins in, there is a part where Ifrit wakes up from bed. When that scene starts, it seems like one of the high pitched notes hangs and gets louder and louder to the point where I had to dash for my remote to mute the TV because it felt like it was about to blow out my eardrums. This continues for about a minute until the explosion happens that stops the background music. I do not beleive this happens on Higan."

Then Kismet responded with a confirmation of the issue:
"I can confirm seeing it on the SD2SNES on 4.4 SNT firmware. It is so loud I had to take my headphones off. Though I'm almost sure it's a bug in the game (I played the translation and it was there.)
 
On the real SNES, translated patch or not, it does not happen.
 
Super NT 4.4 (Arabian Nights - Sabaku no Serei Ou) : Starts at 2:10 minute mark and continues to the 3:40 mark"

https://mega.nz/#!EuIwgabZ!eA2RBVTUqiGRO1uYQzrBZlwbQnscJT0fCnhxus580yg (video demonstrating the issue)

frederic-mahe, 2018-03-08:
@cacophony1 I took the liberty to add some markup to your initial comment. Thanks for reporting this bug here.

SmokeMonsterPacks, 2018-03-26:
This has been fixed in v4.4

closed 2018-03-26 by SmokeMonsterPacks