SmokeMonsterPacks/Super-NT-Jailbreak #59: Glitches: Yoshi's Island PAL (Europe) "Gake Pak Error" with certain settings

SmokeMonsterPacks, 2018-03-09:
## Expected behavior and actual behavior
Yoshi's Island PAL cartridge crashes with the Game Pak Error (like in issue #17)

However it does boot correctly when the SNT is set to PAL system and 720p60 or 1080p60. 

Using 1080p50, 480p60, etc result in the blinking boot error "This game PAK is not designed for your Super Famicom or Super NES."

## ROM name and hash value
European region PAL cart with EN, FR, DE languages

## firmware version
Offiicial v4.4

Error first reported by Dagnus in Discord. Confirmed by Munky.

SmokeMonsterPacks, 2018-03-09:
I'm simply transferring this problem from Discord. Munky also wrote:

> 720p60 works, 1080p60 works, 480p60 gets the game pak screen
actually gimme a second and i'll double check every combo
it seems like the system is defaulting to ntsc when i change the resolution sometimes(edited)
thats whats probably causing it
ok now i'm not sure. cant seem to reproduce it now.
right. think i found it.
my system settings are saved as ntsc with 1080p60
when i power on the system, menu to hardware and change it to pal then change the resolution to 1080p50 after that, it switches the hardware back to ntsc

don't know if that's the problem Dagnus was having, but that seems like a bug to me. changing between 1080p50 and 1080p60 in other circumstances doesnt change the hardware region settings

SmokeMonsterPacks, 2018-03-09:
This may simply be a combination of #17 and #55. If someone with an SNT could clarify, I will decide if this issue should be considered a duplicate and closed.

roflcopter777, 2018-03-11:
I thought GSU-2 games didn't work with ROM loading?

frederic-mahe, 2018-03-11:
@roflcopter777 no ROM loading here, the glitch was observed while playing with a PAL cartridge of _Yoshi's Island_.

roflcopter777, 2018-03-11:
My  mistake, sorry >.>

rklaver, 2018-03-12:
I tried this and I'm not sure what the steps to reproduce are, but I can confirm the following:
- Having the system on the NTSC setting shows the error message (I believe I tried this on 1080p60 and 1080p50)
- Setting the system to 1080p50 and PAL makes the game run
- Saving the 1080p/50 & PAL settings and then rebooting works, game runs
- Changing the resolution from 1080p50 to 1080p60 automatically changes PAL back to NTSC. This may be intended behaviour.

It may be the last issue that is causing confusion. In that case https://github.com/SmokeMonsterPacks/Super-NT-Jailbreak/issues/55 would probably resolve that.

Let me know if this needs more testing.

ShinKaz, 2018-03-14:
Yeah, I'm sorry, I fail to understand OP's issue.

I have a same cart, and it works as intended on my end, just like @rklaver said.

closed 2018-03-14 by SmokeMonsterPacks