SmokeMonsterPacks/Super-NT-Jailbreak #60: Terranigma video glitches, audio dropouts, NTSC patch ineffective

treyturner, 2018-03-09:
## Expected behavior and actual behavior
Expected:
- No video glitches
- Continuous audio
- NTSC patch is effective (so that I can play at 60hz and without switching to PAL mode)

Actual:
- Video glitches in the upper 1/4 of the screen while scrolling (seems OK inside rooms)
- Audio dropouts every several seconds
- Whether or not I use the NTSC patch, I get a copyright error ("This game pak is not designed for your Super Famicom or Super NES")

## Steps to reproduce the problem
Regarding NTSC Patch, reproduction steps are:
Option 1: Boot Terranigma (Europe).sfc in NTSC mode
Option 2: Boot Terranigma (NTSC Patched).sfc in NTSC mode

Regarding video & audio glitches:
Start the game, talk to the elder, exit the house and travel around several screens, especially left and right.

## ROM name and hash value
Terranigma (Europe).sfc 93BA50D853E98E1CA227A2CA72389C0E3AC18D6B50C946B3F618C16C2D3EDD38

The NTSC patch is by BFeely, and is the only one I could find. It's available at https://www.romhacking.net/hacks/541/ among other sites, but when applied, produces ADC2359BAD752932A5E35061F3FA762D47E46EF18E6A75621C13FC5DC73C4A5C, which I don't see on the verified list despite there being a Terranigma (NTSC Fix).smc. Couldn't find that version.

## SD card model
Toshiba 2GB SD-M02G
SD card is pretty old school TBH, but I asked in the community chat and someone else reported problems as well, so I was asked to raise this issue

## Jailbreak firmware version: 
v6.5

## Relevant Super Nt settings
1080p

frederic-mahe, 2018-03-09:
thanks @treyturner , let's see if someone can confirm that bug.

scotthammack, 2018-03-09:
I've only been able to make the visual glitches happen by setting the Super Nt to 1080p60 and PAL mode. However, my version of the NTSC fix ROM is different from the OP's.

Using the NTSC patched ROM, with the following Super Nt settings (applied before loading the ROM):
- 1080p60, NTSC mode: no glitching
**- 1080p60, PAL mode: glitching**
- 1080p50, NTSC mode: no glitching
- 1080p50, PAL mode: no glitching

Using the European ROM:
- 1080p60, NTSC mode: game refuses to boot (region lock screen), as expected
**- 1080p60, PAL mode: glitching**
- 1080p50, NTSC mode: region lock screen
- 1080p50, PAL mode: no glitching

I did not notice any audio dropouts during any of these tests.

SHA256 hashes: Terranigma (NTSC Fix).smc: 03318a9b1039420c567cf73e5edd9ba509cd17e1a049085dafd82dcfce8bc2bc
Terranigma (Europe).sfc: 93ba50d853e98e1ca227a2ca72389c0e3ac18d6b50c946b3f618c16c2d3edd38

treyturner, 2018-03-09:
I came back to close this, because I discovered I was still set to PAL mode apparently (sorry!). Using the 'pure' NTSC patch provided by Great Hierophant (https://cdn.discordapp.com/attachments/418977246842454026/421546816056721419/Terranigma_NTSC.ips) and running the system in NTSC mode, the glitches are gone. I'll leave it up to contributors whether or not this should be help open; my issue is resolved.

closed 2018-03-13 by SmokeMonsterPacks