SmokeMonsterPacks/Super-NT-Jailbreak #62: Reword instructions in README

soegaard, 2018-03-09:
The README says:
> The console LED will turn red and flicker while the firmware is flashed. This process may take a few 
> minutes. The main menu will load when it finishes. Delete the firmware file from your SD card after 
> flashing.

The problem is that the process is not over, when the LED stops blinking.
Therefore you get the impression that something went wrong.

Instead of "few" write the actual number(or add, say, 1) of minutes.

SmokeMonsterPacks, 2018-03-09:
Thanks for the suggestion. I've updated the README to try to make the process more intuitive. I can't add an exact time for the flashing process, since it can vary based on the speed of the SD card used.

closed 2018-03-09 by SmokeMonsterPacks

soegaard, 2018-03-09:
> The main menu will load when it has finished flashing.

This sentence should mention, that there is a considerable pause between the finished flashing and the appearing of the main menu.

SmokeMonsterPacks, 2018-03-09:
OK, I've fixed it. I think it's perfect now. Please let me know what you think.