SmokeMonsterPacks/Super-NT-Jailbreak #63: Glitch: Wizardry Gaiden IV (J) T+Eng v1.01, game freezes after logo, loads after reboot

treyturner, 2018-03-10:
## Expected behavior and actual behavior
Expected: Game boots normally

Actual: Game boots to Ascii Soft logo, then goes to a black screen and freezes. Upon resetting the console via hotkey or physical button-press, the game continues loading after the logo displays.

## Steps to reproduce the problem
Boot the game. Notice the black screen freeze after the logo. Reset the console via hotkey or button-press. Notice the game loads to the title screen.

## ROM name and hash value
Wizardry Gaiden IV (J) T+Eng v1.01 AgentOrange DDS Helly MrRichard.sfc 0789A589CAB4D72E2B7EDD2B1F8038B3E716C97FE03288C38F177F372310B0CE

## Firmware version
v6.5

## SD card model
Toshiba 2GB SD-M02G

## Jailbreak firmware version
v6.5

## Relevant Super Nt settings
1080p/60

SmokeMonsterPacks, 2018-03-10:
Thanks for the report. Could you (or someone reading this) please test this translation on a real SNES to compare? Also, could you please test the original, untranslated ROM on the SNT?

treyturner, 2018-03-11:
Unfortunately the Super NT is the only hardware I have. The untranslated ROM boots normally on it however:
```
Wizardry Gaiden IV - Taima no Kodou (Japan) (Rev A).sfc
SHA-256: BD5023F7FBF1E1D1C5C6B350506B7F727B4A8EC05ADD7281FFAF8F3016FD7099
```

SmokeMonsterPacks, 2018-03-12:
No problem @treyturner. MarcFBR on Discord just confirmed it.

gnadenlos, 2018-03-19:
Works fine from SD2SNES on firmware 4.4.

SmokeMonsterPacks, 2018-03-27:
Closed until someone tests in JB v6.6. Please comment if it needs to be re-opened.

treyturner, 2018-03-31:
Same issue applies in v6.6.

treyturner, 2018-04-23:
Would it be possible to re-open this one? I'm still seeing the same issue.

SmokeMonsterPacks, 2021-02-24:
I'm closing this because it's an unofficial romhack and there will always be problems with them.

closed 2021-02-24 by SmokeMonsterPacks