SmokeMonsterPacks/Super-NT-Jailbreak #64: Gamma boost missing

nwehrle, 2018-03-10:
## Expected behavior and actual behavior
Expected: Gamma boost is shown after scanlines selection
Actual: Gamma bost entry is missing

## Steps to reproduce the problem
Enter the scanline configuration

## ROM name and hash value
not relevant

## firmware version and SD card model

- Jailbreak firmware version: 6.5
- relevant Super Nt settings: 720p60 normal scanlines

Videodr0me, 2018-03-10:
Wasn't Gamma moved to the colors menu with Firmware 4.4/ JB 6.5. Please Check! Its no longer ON/OFF but a slider.

nwehrle, 2018-03-10:
Yep, consulting https://support.analogue.co/hc/en-us/articles/360000557452-Super-Nt-Firmware-Update-v4-4 , it has been changed in v4.3. Thanks for your hint

closed 2018-03-10 by SmokeMonsterPacks