SmokeMonsterPacks/Super-NT-Jailbreak #65: My Super Nt is Bricked... Nothing coming out of the HDMI to TV

somanymatts, 2018-03-10:
Yeah... so I just had to make an account because I thought I followed the instructions and it seemed super easy but now turn my Super NT on and it turns on but nothing is outputting to the TV... talk about a bag of disappointment.  Feeling like YouTubers are just scamming people... what happened... also the whoe GitHub site is a bit bugged... sometimes the search bar just straight up does not work and I have to close the window and redo a google search to get to the page in question... so what is MD5 Checksum application... it's just there and none of the documentation tells me to do anything with that file... is that why my Super NT was bricked.

I am not a programmer but I have a computer science minor from NYU and can program like stupid little Java programs... I'm not going to spend the time training my brain for some formatted bug report, all this info is here.  

Please help.

(template targeting issues with ROMs, feel free to adapt the structure below to your actual problem)

(Do not submit bugs if you are using any means of powering your Super Nt other than the OEM power supply. Only use the original PSU and USB cable when bug testing)

(Please be ready to retest your issue if the firmware is updated)

(Only submit reports for the latest Jailbreak or Official Firmware)

## Expected behavior and actual behavior

## Steps to reproduce the problem

## ROM name and hash value

On Mac and Linux, use command line tools to get the hash values of your ROM (for example: `sha1sum`, `md5sum`). On Windows, you can use the [FCIV](https://support.microsoft.com/en-us/help/889768/how-to-compute-the-md5-or-sha-1-cryptographic-hash-values-for-a-file) utility.

If applicable, check against byuu's [SNES dump database](https://preservation.byuu.org/) and indicate if the bug occurs when using an original cartridge as well.

## firmware version and SD card model

- Jailbreak firmware version: 
- relevant Super Nt settings

somanymatts, 2018-03-10:
Ok... the ReadME file is not accurate... someone needs to like, think a little more logically and rewrite that... you have to wait minutes with the led light no longer flashing with no signal coming out of the hdmi.... come on people... I know instructions are hard to write but like, this is not that complicated to communicate clearly...

MP2E, 2018-03-10:
I suggest you dial your entitlement back 3 notches. We are not Analogue support. It's not our fault you're incapable of reading. Directly from the readme:

"When flashing the LED will turn red and flicker, followed by a considerable pause. The main menu will load when it has finished flashing. Do not power off until the menu loads."

This is perfectly clear.

closed 2018-03-10 by SmokeMonsterPacks

somanymatts, 2018-03-11:
Okay, I understand you may write in code and communicate with machines more than humans but it would actually be more clear if you explained that the signal out to the TV with go away, and to stay calm, when a signal returns and you see the new menu, you know you are done.  Most humans do not read "pause" and assign as much meaning into it as you assume.  

Your entitlement comment was lost on me.  I kinda want to live in a world where humans have learned to communicate well to each other.  But I am aware that humans are only capable of so much at a time.  I can not program what was programmed here and now that all works I am very greatful of the skills used to accomplish this.  Maybe dial the snowflake meter down a bit.  I heard there is some "snowflake" name calling now.  It doesn't really seem like an insult though ;)

SmokeMonsterPacks, 2018-03-11:
So, you really "bricked" your SNT? Are you sure that you're communicating your problem correctly? ;)

somanymatts, 2018-03-11:
It was a brick.  I just reflashed the thing and left the room and did other stuff.  I did sit in front of the tv with the brick loaded with sd roms, confused why my tv said no signal.  Enjoying the snes life now.

frederic-mahe, 2018-03-11:
@somanymatts @MP2E please no name calling.

Writing good documentation is hard, because it is hard to put yourself in the shoes of someone less knowledgeable on the topic at hand. Thanks to @somanymatts 's report, and thanks to @MP2E and @SmokeMonsterPacks, we have a better README now.

soegaard, 2018-03-11:
@somanymatts   Good to hear you got it working. As an aside "bricked" is not something temporary. Bricked means that the machine is beyond repair.