SmokeMonsterPacks/Super-NT-Jailbreak #66: Glitches: Chrono Trigger MSU1 (Hack) Menu Flicker

SmokeMonsterPacks, 2018-03-11:
Clip of problem: 
https://clips.twitch.tv/ClumsySullenTortoiseOSfrog

## Steps to reproduce the problem
Move up/down in the character's menu

## ROM name and hash value
Chrono Trigger Symphony (Blake Robinson)
chrono_msu1.sfc
SHA-1 ef611721787f0984df1e4b6690de530df18dc072

## firmware version and SD card model
- Jailbreak firmware version: v6.5

SmokeMonsterPacks, 2018-03-11:
Closing until verified. I'll reopen it if it can be recreated.

frederic-mahe, 2018-03-11:
Maybe we are more likely to get a confirmation of the bug if the issue remains opened?

SmokeMonsterPacks, 2018-03-19:
Kev has a fix for Chrono Trigger coming in the next official firmware update.

SmokeMonsterPacks, 2018-03-27:
Closed until someone tests in JB v6.6. Please comment if it needs to be re-opened.

closed 2018-03-27 by SmokeMonsterPacks

qwertymodo, 2018-11-25:
Just chiming in to say that the issue is a confirmed bug in the MSU-1 hack that is exacerbated by the SD2SNES in-game hooks (so you can reduce the issue quite a bit by disabling those).  I've further reduced the issue in recent versions of the MSU-1 patch.  It probably won't be able to be eliminated entirely, but I should be able to reduce it to the point where only a couple of scanlines at the top of the screen flicker, rather than the really bad full-screen issue that currently exists in certain areas with the in-game hooks enabled.