SmokeMonsterPacks/Super-NT-Jailbreak #68: Improvement: Screen Boundary Box

B9KnM8UqQ2W2Vwk7uvZ7, 2018-03-11:
**REQUESTED FEATURES**
An option to set a simple white outline on the very edge of the configured display area to assist in people setting up a custom display area. Maybe the outer 1 pixel, 2 pixels or 3 pixels could be solid white.

**REASON**
Different games use different amounts of the available display area to display graphics, quite a few games can have black borders around the horizontal and/or vertical display area. This can make it really awkward to configure the screen dimensions as this varies so much from game to game. One game that may reach the top and the bottom of your TV screen could then cause overscan in the next game you play.

Having the Screen Boundary Box will allow people to always see the outer limits of the display area irrelevant of the game they're configuring it for.

**WHICH GAMES**
It's not a particular game or series of games, some just appear to render fewer pixels than others, perhaps to aid performance.

**FIRMWARE VERSION**
Not in Official v4.4
Not in Jailbreak v6.5

B9KnM8UqQ2W2Vwk7uvZ7, 2018-03-13:
I've added an amendment to this, it's about the initial request and adding an additional option to it rather than it being a completely new request.

SmokeMonsterPacks, 2018-03-28:
Closing minor feature requests to shift the focus back on bugs.

closed 2018-03-28 by SmokeMonsterPacks

kylophone, 2018-03-30:
Doesn't the 240p Test Suite already do something like this?