SmokeMonsterPacks/Super-NT-Jailbreak #69: Tiny Toon Adventures: Buster Busts Loose - Game stops working

foxhoundl, 2018-03-11:
Stage 6 (space) does not load and you get stuck on black screen. When I complete stage 5 (looniversity football match) 1st attempt: after completing bonus game it got stuck when loading stage 6 intro. 2nd attempt: after spinning the bonus wheel, game got stuck when loading bonus game. 3rd attempt: same as 1st attempt.

You must play the game from the beginning only. If you use a password (stage select) the issue will not be reproduced.

Tiny Toon Adventures - Buster Busts Loose! (USA).sfc
SHA1: 04bf92b242ba965c453b6eb13d522afb7db7996b

- Jailbreak firmware version: JB6.5

frederic-mahe, 2018-03-11:
This is a valid ROM:
```
Tiny Toon Adventures - Buster Busts Loose! (USA).sfc
    sha256: ba679a11264e9695895a6c17358a41e8459be06166d056811df9c2738fef3d0d
    sha1: 04bf92b242ba965c453b6eb13d522afb7db7996b
    md5: 68e83fe143a01ea5a68bbdfc17a34a2d
    crc32: ab48d27a
```
Ideally, we would need someone to confirm the bug with the ROM, and we would need to know if the bug occurs with a cartridge version of the game too.

SmokeMonsterPacks, 2018-03-12:
Glitchez at Discord just posted:
> On the jb I started up a game and before the first level intro I got black screened. Tested it five more times and no issue. Then put it an sd2 and 5 times no issue. Got a blackscreen playing it on the SD2SNES after a level 2 death. So the bug appears on a level load screen and after a death.

dreimer1986, 2018-03-16:
This game seems to be a lil troublemaker... I tried the PAL version from no-intro validated and can play the game rather well, except that the spinning wheel is shown twice and one about 20% moved to the right and flickering and that the sound starts to slowdown badly after the first wheel cutscene and never comes back to normal when you continue to play. I can verify later on SD2SNES and maybe even with original cart (if still in my collection...)

BellyBel, 2018-03-16:
Just wanted to confirm that this bug also affect the Japanese cartridge version. I had severe audio slowdown on music and effects after a death transition at the end of level 2.

dreimer1986, 2018-03-17:
Sound bug confirmed be gone when game used with SD2SNES. Spinning wheel only is buggy on 60Hz output. (PAL game is what I use, dont remeber that I had this problem on a real SNES, but can be ignored I would say)

dododub, 2018-03-17:
The slowdown happens on original NTSC cart as well. Started immediately after I finished the first level. At first the first level wouldn’t even load at all, after a reset it did. Cleaned cart.

dododub, 2018-03-17:
And just be complete: on 4.4 firmware, NTSC mode.

dreimer1986, 2018-03-17:
Then this bug happens not everytime. I had no problems right now, but at the very beginning mentioned above. (PAL GB one)

SmokeMonsterPacks, 2018-03-19:
Kev has a fix for Tiny Toons coming in the next official firmware update.