SmokeMonsterPacks/Super-NT-Jailbreak #70: SMW Romhack: The Joy of Kaizo 1.5.1 not saving

kickinkame, 2018-03-12:
## Expected behavior and actual behavior
If i return from the game to the menu i should get a save dialog and be able to save the game. Works fine in vanilla Super Mario World. Doesent work with this Romhack.

Tested in Higan with the same ROM. Saving works fine there. I dont have a flashcard to test this on the original firmware.

## Steps to reproduce the problem
Start the game.
Finish a level.
Return to the menu (select+down).
No save dialog is shown.
select doesent return back to the game.
if you go out of the sd card menu (start) you can go back into the game with select. But the Menu will randomly open again.

## ROM name and hash value
https://www.smwcentral.net/?p=section&a=details&id=16866

Patch: JOY OF KAIZO 1.5.1.bps 
sha1: 55ef1f4ff57d95e4b6082fbc7f7c623ee167a229

Rom: JOY OF KAIZO 1.5.1.sfc 
sha1: 3a3f00f5c7459f16312a723dd5e0a4df7b398590

## firmware version and SD card model

- Jailbreak firmware version: JB6.5
- relevant Super Nt settings: NTSC, 1080p60
- sdcard: SanDisk Ultra SDHC I 16 GB

SmokeMonsterPacks, 2018-03-13:
Could someone please test this on a real SNES? Since it's a SMW Hack, I wouldn't be surprised if it's just coded incorrectly. If they only tested their work in Higan, it would explain why saving might work in emulation but not real hardware.

SmokeMonsterPacks, 2018-03-13:
I'm closing this since it's probably a problem with the hack.

kickinkame, 2018-03-22:
@SmokeMonsterPacks i asked someone to test this on real hardware with a flashcard, and saving seems to work there.

here is a video: https://www.twitch.tv/videos/241627225?t=3h55m10s

i also did some research. There is a patch applied called 'SRAM Plus' https://www.smwcentral.net/?p=section&a=details&id=14762

in the readme it states: 'This patch basically rewrites all of the SRAM saving, loading, and erasing save file routines that SMW uses. It uses DMA to copy the addresses to and from SRAM, meaning that it is probably as efficient as possible. The complicated checksum check and direct file copy was replaced with simple validation bytes, meaning that you have more SRAM at your disposal. The patch also frees up 141 bytes at $1F49 by moving the SRAM buffer to $1EA2.'

could this probably be the problem?

thx for any help in advance

kickinkame, 2018-03-24:
@SmokeMonsterPacks @frederic-mahe 

i tested 3 more hacks with the 'SRAM PLUS' patch applied. Saving didnt work in any of them, so this seems related. 

This patch is pretty common, as it is needed to turn the live counter into a death counter.

As all of this games do work fine on original hardware with a flash card. 
I think this needs to be reopend.

frederic-mahe, 2018-03-25:
I renamed the issue (title is too long though, suggestions welcomed)

SmokeMonsterPacks, 2018-03-27:
Closed until someone tests in JB v6.6. Please comment if it needs to be re-opened.

closed 2018-03-27 by SmokeMonsterPacks

kickinkame, 2018-03-27:
tested on the latest firmware (6.6) saving works fine.

thx for fixing